bluesky/ophyd-async #396: ADAravis does not support internal triggering

callumforrester, 2024-06-18:
`ADAravis` detectors cannot be `count`ed as they don't support internal triggering:

```shell
ERROR:blueapi.cli.cli:<AsyncStatus, task: <coroutine object StandardDetector.trigger at 0x7feb33ea7840>, errored: ValueError("AravisController only supports the following trigger types: (<DetectorTrigger.constant_gate: 'constant_gate'>, <DetectorTrigger.edge_trigger: 'edge_trigger'>) but was asked to use DetectorTrigger.internal")>

```

olliesilvester, 2024-06-27:
I ran into this issue as well. For my work-around, I made the `Aravis` detector device `override` trigger, so that it sets the trigger type to `constant_gate` by default

coretl, 2024-06-27:
This is a simple bug:
https://github.com/bluesky/ophyd-async/blob/f8009fdd6b8258bcbc4bf7a69e342a03994c42f9/src/ophyd_async/epics/areadetector/controllers/aravis_controller.py#L62-L72

Should have `DetectorTrigger.internal` in `supported_trigger_types`

PR #420 (2024-06-27): Add DetectorTrigger.internal to AravisController supported trigger types
Fixes #396
closed 2024-06-27 by olliesilvester