bluesky/ophyd-async #405: Decide the interface between `StandardDetector` and `DetectorController` w.r.t. the `TriggerInfo` that is passed down in `prepare`

coretl, 2024-06-19:
As mentioned in https://github.com/bluesky/ophyd-async/pull/401#discussion_r1644527435 and https://github.com/bluesky/ophyd-async/pull/401#discussion_r1644506670

coretl, 2024-06-19:
@callumforrester @abbiemery @evalott100 @DiamondJoseph thoughts?

callumforrester, 2024-06-19:
I think there is a use case for keeping the detector to minimum deadtime while the _system_ as a whole has a longer deadtime, which is what this diagram covers. Do any your permutations cover that?

[image]

coretl, 2024-06-19:
Yes, as this is a property of the trigger generation system, rather than the detector. In the case above the `TriggerInfo.deadtime` (which the docstring describes as the minimum deadtime) will be the output of `get_deadtime(exposure)`, while the trigger system will actually provide a bigger gap between triggers as shown by your diagram

coretl, 2024-07-25:
Need a discussion with @jwlodek and @tomtrafford

ZohebShaikh, 2024-09-10:
After future discussion,

We have decide to have 

```
StandardDetector.prepare() -> DetectorController.prepare()

StandardDetector.kickoff() -> DetectorController.arm()

StandardDetector.trigger() -> DetectorController.arm()

```

PR #568 (2024-09-13): Interface change of StandardDetector and Standard Controller
closes #405 #542 https://github.com/bluesky/ophyd-async/issues/549
closed 2024-09-13 by ZohebShaikh