bluesky/ophyd-async #410: Allow for alternative naming strategies for child devices

DiamondJoseph, 2024-06-25:
We have a few Devices that should be transparent to child naming.

E.g. where `saxs` is an instance of `PilatusDetector`: `saxs-drv-acquire_time`, where `saxs-drv` must be a device to allow its signals to be connected automatically when `saxs` is connected, but we do not want the field to be named `saxs-drv`. This was also observed when attaching a `_metadata_holder` device as a child of `saxs` that should provide SoftSignals for processing.

Suggestion: devices that are underscore variables on their parent name their children as though their name was their parents' name?

So that 

```python
class NXSasPilatus(PilatusDetector):
  _metadata_holder: MetadataHolder

md(pixel_size_x=(0.2, "mm"))
saxs = NXSasPilatus(md)
saxs.read() == {
  "saxs-pixel_size_x": 0.2
}
```

olliesilvester, 2024-06-25:
This came up for MX when creating `DeviceA` which has attributes `SignalA` and `MovableA`. `MovableA` is a class which also uses `SignalA`. This means that after calling `DeviceA super().__init__`, `SignalA` is named like `'DeviceA-MovableA_SignalA'`, when it makes more sense for it to be `'DeviceA_SignalA'`

coretl, 2024-06-25:
> Suggestion: devices that are underscore variables on their parent name their children as though their name was their parents' name?

I would prefer if devices with underscore variables were skipped from naming completely.

@olliesilvester if we apply this, and you set `MovableA._sig = SignalA` then it would be named according to `DeviceA` reference, would this work for you? Although thinking about it, why are `MovableA` and `DeviceA` separate objects in the first place? Could the be merged into one?

@DiamondJoseph for the metadata holder, could you name it with an underscore, then name it the same as the parent?
Prior art:
https://github.com/bluesky/ophyd-async/blob/9165c5ae811074b92a0d549946fe85915c538c7a/src/ophyd_async/epics/motion/motor.py#L48-L51

olliesilvester, 2024-06-25:
> @olliesilvester if we apply this, and you set `MovableA._sig = SignalA` then it would be named according to `DeviceA` reference, would this work for you? 
> 
Yes that should work for us
>Although thinking about it, why are `MovableA` and `DeviceA` separate objects in the first place? Could the be merged into one?
>
In this case, doing a `set` on `DeviceA` does a different thing to a `set` on `MovableA`, but they move use the same signal for it

DiamondJoseph, 2024-07-04:
> @DiamondJoseph for the metadata holder, could you name it with an underscore, then name it the same as the parent? Prior art:
> 
> https://github.com/bluesky/ophyd-async/blob/9165c5ae811074b92a0d549946fe85915c538c7a/src/ophyd_async/epics/motion/motor.py#L48-L51

It's the signals nested under _metadataholder that want to be renamed, not the _metadataholder itself.

coretl, 2024-11-22:
@DiamondJoseph how urgently does this need to be fixed? Shall we discuss on Monday?

DiamondJoseph, 2024-11-25:
I think Oli's use case more urgent than mine- currently there is handling for the current naming in the NeXus service and the softsignals we're utilising are planned to be replaced with injecting that MD into documents from an external service.