bluesky/ophyd-async #416: Pilatus timestamps link/value is not correct

DiamondJoseph, 2024-06-26:
Pilatus seems to have 2 "timestamp" fields, /entry/instrument/NDAttributes/NDArrayTimeStamp, which contained N values of (1088256331) (Timestamp 1088256331 corresponds to: 2004/06/26 14:25:31 (?))

coretl, 2024-06-26:
This is https://github.com/areaDetector/ADPilatus/pull/18

@jsouter what's the status of this on I22?

jsouter, 2024-06-27:
@coretl Looks like the version of ADPilatus being used on I22 is 2-9dls4 (from March 2022), there's a 2-9dls5 but don't think it addresses any timestamping issues. 
https://github.com/areaDetector/ADPilatus/commit/5a649ceb3392b581a19c005db6584fde73ab6ded
This is the change Mark pushed through, I can pull that onto dls-master for ADPilatus and make a new release with it

coretl, 2024-07-22:
@jsouter do you know what is deployed on I22 now?

jsouter, 2024-07-22:
> @jsouter do you know what is deployed on I22 now?

CS16 and CS17 are both using the new release of ADPilatus with the timestamp changse

coretl, 2024-07-22:
Closing as this should now be deployed

closed 2024-07-22 by coretl