bluesky/ophyd-async #444: TypeChecking for AreaDetector StandardDetectors

DiamondJoseph, 2024-07-10:
When given a sequence of StandardDetectors, that are compatible with a plan, we may want to configure plugin chains for all AreaDetectors in that sequence. Currently a possible check is checking `isinstance(detector.drv, ADBase)`: after #213 will this still be the proscribed method? the availability of the driver from the detector is not guaranteed by the StandardDetector definition.

coretl, 2024-07-10:
We need to associate particular (explicitly mentioned) plugins with the scalar datasets they produce as per https://github.com/bluesky/ophyd-async/issues/282 but I don't think we need to do anything with walking the whole chain. Is this what you are referring to?

DiamondJoseph, 2024-07-11:
Just configuring explicit plugin(s) for what was needed to get the Linkam plan for i22 working again (enabling StatsSum)

coretl, 2024-07-12:
ok, so I suspect #282 is what we need then rather than this. Would you agree?

DiamondJoseph, 2024-07-12:
If I have a group of StandardDetectors, `foo`, `bar` and `baz`: `foo` and `bar` are backed by AreaDetectors, while `baz` is a non-AreaDetector. I want to configure all of my detectors that have the ADStatsPlugin to calculate, save and expose the signal for StatsSum. Isn't #282 just how that signal then propagates into documents? I need an easy way to do the following, that shouldn't rely on `hasattr` and has typing support.

```python
for det in detectors:
  if hasattr(det, "stats"):
    configure_sum(det)
```

coretl, 2024-07-15:
Closing in favour of #282 as we are likely to set this up programmatically in a plan stub at the beginning of the scan, where we have knowledge of the Detector instance type, and therefore the fact its plugins are `NDArrayBase` and can set `NDAttributeXML`, rather than in the generic scan where we only know it's a `StandardDetector`

closed 2024-07-15 by coretl