bluesky/ophyd-async #447: ophyd_async.core.signal.set_and_wait_for_value() within tolerance

kathryn-baker, 2024-07-11:
Hello, I am interested in understanding more about how we can use `ophyd-async` to improve the way we interact with our EPICS PVAccess control system. 

I have created a test EPICS RW signal connected to a SET/READ pair of PVs where I expect there to be some delay between the updates:

```python
from ophyd_async.epics.signal import epics_signal_rw
from ophyd import Component, Device, EpicsSignal, EpicsSignalRO
import asyncio
from ophyd_async.core.signal import set_and_wait_for_value

async def main():
    signal = epics_signal_rw(
        float, read_pv="pva://TEST:PV:READ", write_pv="pva://TEST:PV:SET"
    )
    connection = await signal.connect()
    status = await set_and_wait_for_value(signal, 8, timeout=1)

asyncio.run(main())
```

Currently the read PV is configured to have some noise associated with it (as we might expect to see from the hardware). However this means that I get a TimeoutError on the `set_and_wait_for_value()` call because it expects the values to match exactly. 

```bash
TimeoutError:  didn't match 8 in 1s, last value 8.001638609114593
```

Is there any option to provide a tolerance for this command to say that I want it to update to within 0.01 or something? For example:
```python
status = await set_and_wait_for_value(signal, 8, timeout=1, tolerance=0.01)
```

I'm currently using `ophyd-async==0.3.4` with Python 3.10.14

coretl, 2024-07-12:
There isn't at the moment, but this would be a fairly easy addition. It would need adding to `wait_for_value` and the tolerance passing down from `set_and_wait_for_value`. I should note that we are about to do a repository restructure in #279 , so these functions will be moving to a different file in about a week's time, so these changes should probably be done after that ticket is complete.

Is this something you'd like to make a PR for, or shall we put it on our backlog?

kathryn-baker, 2024-07-12:
Thanks for the explanation, I think with that I should be able to try doing a PR myself. I'll wait until the restructure goes through though.

kathryn-baker, 2024-07-15:
With more digging I also found that you could solve this issue through the use of a callable in the `match` parameter for the `wait_for_value()` call in the following way:

```python
    signal = epics_signal_rw(
        float, read_pv="pva://TEST:PV:READ", write_pv="pva://TEST:PV:SET"
    )

    set_val = 9
    tolerance = 0.1

    def within_tolerance(v):
        return (v > set_val - tolerance) and (v < set_val + tolerance)

    signal.set(set_val)
    status = await wait_for_value(signal, within_tolerance, timeout=2)
```

Although this requires you to separate the `set()` call from the `wait_for_value()` call so I think it would still be nice to have the tolerance parameter as an option.

DominicOram, 2024-07-16:
Alternatively we could change `set_and_wait_for_value` to also take a `Union[T, Callable[[T]]`? Seems more flexible and gives parity with the behaviour of `wait_for_value`

coretl, 2024-07-16:
Yes, I think that makes sense. Let's make `wait_for_value` and `set_and_wait_for_value` both take a `Union[T, Callable[[T]]`, and say that you do tolerance via the callable.

closed 2024-07-16 by coretl