bluesky/ophyd-async #448: Create a PandA flyer that uses a SEQ table to do time based triggers within a position based gate

coretl, 2024-07-11:
Want something similar to https://github.com/bluesky/ophyd-async/blob/5474cf4210c15994854fc29f77a2842212b1c1ea/src/ophyd_async/panda/_trigger.py#L15-L39 but for a SEQ table implementing the following scheme:

[image]

## prepare
Make a dataclass `MresOffset` that includes the Mres and Offset got from the relevant PandA input.
Take a `List[scanspec.Frames[MresOffset]]`, calculate the entire trajectory (this will later be only the first 4000 points), find the gaps, in each gap insert:
- Wait for BITA=1 for GPIO
- Wait for POSA to exceed the value of the first compare point
- Generate N triggers for the row

At the moment we should fail if there are more than 4096 rows generated

## kickoff
Set `ENABLE=ONE` and wait for `ACTIVE`

## complete
Wait for `ACTIVE` to drop

coretl, 2024-07-25:
[image]

subinsaji, 2024-07-25:
A picture to remind ourselves of the logic and conversations

[image]

coretl, 2024-07-26:
I had some better ideas for structure. We make a `TimeBasedSeqController` where the position compare part is optional. In `prepare()` we optionally additionally take a `PositionCompareInfo` with a `motor: Motor`, `resolution: float` and `offset: float`. If supplied then we put in the lines for the compare, otherwise we omit them. This will give us both GPIO gate and time trig, as well as position gate and time trig.