bluesky/ophyd-async #449: Investigate put to PANDA:PCOMP1:ENABLE in kickoff timing out

coretl, 2024-07-11:


coretl, 2024-07-24:
I cannot reproduce this, but it is suspicious that this field is a `bit_mux` and we are not setting `always_update=True` for this in the IOC, while we are for some other fields. 

Maybe @AlexanderWells-diamond can think of a reason that a pvput to a bit_mux field which has just been changed to a different value a few milliseconds earlier would time out after 10 seconds?

AlexanderWells-diamond, 2024-07-25:
I think I'd need an introduction to what is supposed to be going on here to even being answering the question. Is the first `pvput` waiting for the record to confirm it has finished processing before the next `pvput`? Is the `bit_mux` field we're talking about the one [created here](https://github.com/PandABlocks/PandABlocks-ioc/blob/5309422c27f8fbff2d221b551fe531734c493456/src/pandablocks_ioc/ioc.py#L1166)?

coretl, 2024-07-26:
> Is the first `pvput` waiting for the record to confirm it has finished processing before the next `pvput`? 

Yes, we do `pvput <bit_mux> ZERO` closely followed by `pvput <bit_mux> ONE`, and sometimes the second one times out.

We use the `wait=True` argument to `p4p.Context.put` which I think waits for processing to complete:
https://github.com/bluesky/ophyd-async/blob/c6a9c79743c10ee256b60f34e0801bc70f8e1c4f/src/ophyd_async/epics/signal/_p4p.py#L409-L418 

I guess we should probably test that though...

> Is the `bit_mux` field we're talking about the one [created here](https://github.com/PandABlocks/PandABlocks-ioc/blob/5309422c27f8fbff2d221b551fe531734c493456/src/pandablocks_ioc/ioc.py#L1166)?

Yes

closed 2024-11-12 by coretl