bluesky/ophyd-async #459: Fix docs build

coretl, 2024-07-16:


coretl, 2024-07-16:
@olliesilvester FYI

evalott100, 2024-07-16:
Seems that when the autodoc generates `ophyd_async.epics.rst`:

```
.. note::

    Ophyd async is included on a provisional basis until the v1.0 release and 
    may change API on minor release numbers before then

ophyd\_async.epics
==================

.. automodule:: ophyd_async.epics

   
   
   

   
   
   

   
   
   

   
   
   

.. autosummary::
   :toctree:
   :template: custom-module-template.rst
   :recursive:

   ophyd_async.epics.areadetector
   ophyd_async.epics.demo
   ophyd_async.epics.motion
   ophyd_async.epics.pvi
   ophyd_async.epics.signal
```

The warning is caused by the `ophyd_async.` at the bottom, which doesn't match the other modules - e.g panda
```

   .. autosummary::
      :toctree:
      :template: custom-class-template.rst
      :nosignatures:
   
      CommonPandaBlocks
      HDFPanda
      PcompBlock
      PcompDirectionOptions
      EnableDisableOptions
      PcapBlock
      PulseBlock
      SeqBlock
      SeqTable
      SeqTableRow
      SeqTrigger
      PandaPcapController
      TimeUnits
      DataBlock
      StaticSeqTableTriggerLogic
      StaticPcompTriggerLogic
```

which doesn't include the `ophyd_async`. No idea how to fix this so @abbiemery should probably do her work first.

abbiemery, 2024-07-17:
I have included sphinx and pinned it to allow the docs to build for now. We use sphinx via sphinx-autobuild so once we have solved the problem this can be removed again.

stan-dot, 2024-11-18:
@coretl how to replicate this issue?
when I run `tox -e docs` I get even more errors now

[error.txt](https://github.com/user-attachments/files/17803476/error.txt)

PR #700 (2025-02-11): Add 0.10 docs
Rewrite of the documentation and its structure including:
- Making sim and epics.demo devices do the same
- Placeholder for the same for tango.demo and fastcs.demo devices
- New tutorials, how-tos and explanations
- Reduce installed dependencies

The first half of #721 
Fixes #677 
Fixes #671
Fixes #484 
Fixes #480
Fixes #479
Fixes #459 
Fixes #443 (well enough anyway)
Fixes #386
Fixes #349 
Fixes #303 
Fixes #246 
Fixes #227 
Fixes #193 
Fixes #622
closed 2025-02-11 by coretl