bluesky/ophyd-async #479: from ophyd v1 to ophyd-async (was v2?)

mguijarr, 2024-07-24:
Hi !

Prior to try ophyd-async, and prior to embarking into some heavy development, I am wondering if any
guide / tutorial / documentation / notes exist to tell how to do a transition from Ophyd v1 to the new version.

What are the differences ? (apart from the asynchronous execution model)
What is compatible ?
What is incompatible ?
What has to be changed ?

Thanks in advance for some hints.

mguijarr, 2024-07-24:
Some explanation about "why ophyd-async ?" would be useful, too ; what problem does it solve, that was impossible to solve with the original version ?

coretl, 2024-07-26:
I will answer the questions here now, then leave this ticket open so we can transfer the answers into the docs before we make the 1.0 release later this year.

> What are the differences ? (apart from the asynchronous execution model)
> Some explanation about "why ophyd-async ?" would be useful, too ; what problem does it solve, that was impossible to solve with the original version ?
> What is compatible ? What is incompatible ? What has to be changed ?

This should go on the top level README.

The main differences from ophyd (sync) are:
- Asynchronous Signal access, simplifying the parallel control of multiple Signals
- Support for PVA and Tango as well as the traditional CA
- Better library support for splitting the logic from the hardware interface to avoid complex class heirarchies

It was written with the aim of implementing flyscanning in a generic and extensible way with highly customizable devices like PandABox and the Delta Tau PMAC products. Using async code makes it possible to do the "put 3 PVs in parallel, then get from another PV" logic that is common in flyscanning without the performance and complexity overhead of multiple threads.

Devices from both ophyd and ophyd-async can be used in the same RunEngine and even in the same scan. This allows a per-device migration where devices are reimplemented in ophyd-async one by one.

Then somewhere in the docs should be a how-to guide on reimplementing an ophyd device in ophyd-async...

PR #700 (2025-02-11): Add 0.10 docs
Rewrite of the documentation and its structure including:
- Making sim and epics.demo devices do the same
- Placeholder for the same for tango.demo and fastcs.demo devices
- New tutorials, how-tos and explanations
- Reduce installed dependencies

The first half of #721 
Fixes #677 
Fixes #671
Fixes #484 
Fixes #480
Fixes #479
Fixes #459 
Fixes #443 (well enough anyway)
Fixes #386
Fixes #349 
Fixes #303 
Fixes #246 
Fixes #227 
Fixes #193 
Fixes #622
closed 2025-02-11 by coretl