bluesky/ophyd-async #481: Discuss renaming HDFDataset and HDFFile

abbiemery, 2024-07-24:
Both HDFDataset and HDFFile have names which are not descriptive to what they actually are. It would be helpful to name these appropriately and to go through where they are used to ensure their meaning is clear and consistent.

coretl, 2024-07-26:
How about `HDFDocumentComposer` and `HDFDatasetInfo`? Or some other combination of `Stream`, `Resource`, `Dataset`, `Info`, `Document` and `HDF`?

@DiamondJoseph are you good at naming things?

DiamondJoseph, 2024-07-29:
no

A HDFFile produces 1 StreamResource per HDFDataset right? They all point to data stored in the same place

HDFDataset to me makes sense, or it could be HDFDatakey to map to the ObjectKey/DataKey?
HDFFile -> HDFDocumentComposer makes sense to me

abbiemery, 2024-07-29:
Ok, `HDFDocumentComposer` and `HDFDatakey` it is

coretl, 2024-12-09:
See if `*DatasetDescriber` produces `HDFDatasets` and if so name it `HDFDatasetDescription`. The top level `HDFDocumentComposer` works fine.

PR #794 (2025-02-27): Rename HDFFile -> HDFDocumentComposer
and HDFDataset -> HDFDatasetDescription
remove the incorrect "SWMR" parameter too

Fixes #481
closed 2025-02-27 by coretl