bluesky/ophyd-async #487: Prepare in Motor set negative velocity when travel toward negative direction.

Relm-Arrowny, 2024-07-24:
**Describe the bug**
Prepare in Motor set negative velocity when travel toward negative direction, which cause assert error during kickoff. 

**To Reproduce**
run a plan where end_position <start_position:
```python
        fly_info = FlyMotorInfo(
            start_position=start,
            end_position=end,
            time_for_move=abs(start - end) / motor_speed,
        )
        yield from bps.prepare(motor, fly_info, group=grp, wait=True)
        yield from bps.kickoff(motor, group=grp, wait=True)
```

**Expected behaviour**
No error, when moving toward negative direction.

**Additional context**
[Line](https://github.com/bluesky/ophyd-async/blob/955414ab258a8ef48f624fcefa43c7f2c33e86de/src/ophyd_async/epics/motor.py#L121) that cause the assertion error.
[Function ](https://github.com/bluesky/ophyd-async/blob/955414ab258a8ef48f624fcefa43c7f2c33e86de/src/ophyd_async/epics/motor.py#L180)that calculate the velocity.
[Where ](https://github.com/bluesky/ophyd-async/blob/955414ab258a8ef48f624fcefa43c7f2c33e86de/src/ophyd_async/epics/motor.py#L116) negative velocity could be setted.

Relm-Arrowny, 2024-09-02:
Can I pick this one up? I am running 0.5 on p99 and this is a show stopper for me.

coretl, 2024-09-02:
> Can I pick this one up? I am running 0.5 on p99 and this is a show stopper for me.

yes please

PR #552 (2024-09-02): Stop prepare setting negative velocity
fix #487
closed 2024-09-02 by Relm-Arrowny