bluesky/ophyd-async #502: `SimMotor` is not importable without dev dependencies

callumforrester, 2024-07-31:
I can't do: `from ophyd_async.sim.demo import SimMotor` in a project that depends on `ophyd-async`. 

```shell
File ~/projects/bluesky/bluesky-stomp/venv/lib64/python3.10/site-packages/ophyd_async/sim/demo/_pattern_detector/_pattern_generator.py:4
      1 from pathlib import Path
      2 from typing import AsyncGenerator, AsyncIterator, Dict, Optional
----> 4 import h5py
      5 import numpy as np
      6 from bluesky.protocols import DataKey, StreamAsset

ModuleNotFoundError: No module named 'h5py'
```

h5py is a dev dependency, so it should move into primary dependencies?

coretl, 2024-07-31:
I'm erring on the side of a `ophyd_async[sim]` extra that includes h5py

jwlodek, 2024-07-31:
Should we re-organize imports to allow for importing SimMotor without needing h5py? As I understand h5py is only necessary for the pattern generator sim device. So SimMotor maybe should be importable w/o needing this library?

stan-dot, 2024-11-18:
at the moment the main branch does have the `ophyd_async[sim]` extra

importing SimMotor without the h5py would only make sense if we restrict the methods --> if a method uses the h5py libarary we'd throw an error 

```python
import functools

def require_library(lib_name: str):
    """
    A decorator to ensure a specific library is available in the scope.
    If the library is not present, raises an ImportError.
    Caches the result for subsequent checks.
    """
    library_cache = {}

    def decorator(func):
        @functools.wraps(func)
        def wrapper(*args, **kwargs):
            # Check the cache first
            if lib_name not in library_cache:
                try:
                    # Attempt to import the library
                    __import__(lib_name)
                    library_cache[lib_name] = True
                except ImportError as e:
                    # Cache the failure and raise an error
                    library_cache[lib_name] = False
                    raise ImportError(f"Required library '{lib_name}' is not available.") from e

            # If library exists, run the decorated function
            if library_cache[lib_name]:
                return func(*args, **kwargs)

        return wrapper

    return decorator

```

coretl, 2024-11-18:
I think we should organise `sim` into various subpackages that can each have their own deps, e.g.
```
sim\
  motor\
  detector\
  testing\
```

callumforrester, 2024-11-18:
I'm inclined to agree with @stan-dot, I don't think it is too costly to require the user to install h5py in order to make use of the sim package. Extra dependencies is a nice way to avoid requiring everything for every user but it's possible to go too far in the other direction too. H5py is one library and not too heavyweight, so I'd throw in my vote for requiring it if you want to use the sim package.

coretl, 2024-11-25:
Decision: make `h5py` a sim dependency and keep with `demo`

coretl, 2025-03-28:
There is now a `sim` extra with the `h5py` dependency and a `demo` dependency with `ipython` and friends

closed 2025-03-28 by coretl