bluesky/ophyd-async #519: Correctly handle pending tasks in our tests

olliesilvester, 2024-08-13:
Running unit tests sometimes gives us a few `Task was destroyed but it is pending!` messages. This is because, in our pytests, we either:

1. Use the RE fixture to create an event loop, do the test, then the fixture closes the event loop at the end of the pytest using pytest's `add_finalizer`. This can run before awaiting for pending tasks, so pending tasks can be cancelled.

2. Use an Asyncio event loop outside of the RE and the same as above

In both cases, if we create a task using `asyncio.create_task `and then finish the test before any `await`s are ran, there is a chance that the event loop gets closed with pending tasks, which causes that warning message
To make sure these pending tasks get completed, we can either:

1. Make our event loop fixtures do an await at the end of a pytest, eg
```
def RE(request)
...
if pending:
            loop.run_until_complete(asyncio.gather(*pending, return_exceptions=True))
```

2. Make our event loop fixtures raise an error if there are pending tasks at the end, and tell us which tasks they are

3. Something else clever which I didn't think of

evalott100, 2024-08-13:
Closing https://github.com/bluesky/ophyd-async/issues/260 in favour of this

jwlodek, 2024-08-14:
I think #498 is also a duplicate of this. Suggest we close that since this goes into quite a bit more detail.

evalott100, 2024-08-19:
> That gets my vote. Do you know how we would do that?

We could do this for tests using the `RE` fixture by getting pending.

We could also do something like:
``` python
@pytest.fixture(autouse=True)
async def assert_no_pending_tasks(request):
    # There should be no tasks pending after a test has finished
    try:
        yield
    finally:
        loop = asyncio.get_event_loop()
        pending = asyncio.all_tasks(loop)
        if pending # and some_filtering_perhaps:
            unfinished_tasks = ...
            raise RuntimeError(
                "Not all tasks {unfinished_tasks} closed during test."
            )

        
```

olliesilvester, 2024-08-19:
> We could do this for tests using the `RE` fixture by getting pending from the run engine loop.
> 
Yep, this is what I initially tried in that other PR. It raises errors but it's not necessarily obvious as to which test an unfinished task came from

evalott100, 2024-08-19:
> Yep, this is what I initially tried in that other PR. It raises errors but it's not necessarily obvious as to which test an unfinished task came from

I think we can use `traceback` in the standard library and raise a new `UnresolvedTaskError` encapsulating that info.

PR #538 (2024-09-17): Added a fixtures that checks if tasks are still running in tests
Closes #519
closed 2024-09-17 by evalott100