bluesky/ophyd-async #520: Some tests fail on 3.12

dperl-dls, 2024-08-14:
In testing the new p4p release I came across some test failures which I think are wrong, in particular [`tests/core/test_mock_signal_backend.py::test_blocks_during_put`](https://github.com/bluesky/ophyd-async/blob/d5da999f7f066fd895c9c278c599ec4977bd61d0/tests/core/test_mock_signal_backend.py#L184-L200)

In lines 186 and 187 I believe this should test for `"first_value"` as the signal shouldn't have changed yet with puts blocked - this works correctly on python 3.12 but not earlier versions.

dperl-dls, 2024-08-14:
similarly https://github.com/bluesky/ophyd-async/blob/d5da999f7f066fd895c9c278c599ec4977bd61d0/tests/epics/advimba/test_vimba.py#L32 fails, I believe because [these sets](https://github.com/bluesky/ophyd-async/blob/d5da999f7f066fd895c9c278c599ec4977bd61d0/src/ophyd_async/epics/advimba/_vimba_controller.py#L52-L54) are not awaited - @jwlodek do you think this is correct?

jwlodek, 2024-08-14:
Indeed I believe those should be awaited. It's interesting that that would not have been caught in tests before

dperl-dls, 2024-08-14:
Great. I've fixed these in my python 3.12 branch https://github.com/bluesky/ophyd-async/tree/update_p4p_and_py312 but am waiting on another release of p4p built against numpy 2 for python versions 3.11 and 10 before it can be merged

coretl, 2024-11-22:
@dperl-dls is this still relevant?

dperl-dls, 2024-11-22:
@coretl probably, I guess there's still no 3.12 in CI here so it seems likely