bluesky/ophyd-async #521: Provide a NoOpStatus

DominicOram, 2024-08-15:
`ophyd` has a `NullStatus` (https://github.com/bluesky/ophyd/blob/050b98f34d68bd9577179accfdfca8f635cc0f31/ophyd/sim.py#L51) that immediately returns, this is very useful for testing as it's something you probably want to inject into many devices. We should provide something similar in `ophyd-async` where it basically just does `AsyncStatus(asyncio.sleep(0.0))`. It will also be nice to be able to:
* Be able to provide an identifier (such as the name of the test) so that we can easily identify it later if needed
* Be able to provide a flag that causes the status to be in error

coretl, 2024-08-16:
Can you provide a use case please? The closest we have at the moment is:

```python
@AsyncStatus.wrap
async def kickoff(self) -> None:
    pass
```

Which could be (slightly) reduce if we had a NoOpStatus:

```python
def kickoff(self) -> Status:
    return NoOpStatus()
```

DominicOram, 2024-08-16:
It's more for downstream testing e.g.

```python
class MyDevice(Device):
    def set(self, value):
        # Some long running thing that checks against PVs etc

def test_my_plan():
    my_device.set = MagicMock(return_value=NullStatus())
    my_plan(my_device)
```

In this case the logic in set is too complicated for the mocked object (via `DeviceCollector(mock=True)`) to handle out the box but we also don't want to have to worry about internal details of it in our plan. For this test we just care that it works (or maybe doesn't work hence the requirement on being able to provide an error)

coretl, 2024-08-16:
Would it have to be a subclass of `AsyncStatus` so you can `isinstance` it, or would a factory function work? I'm slightly averse to big nested class heirarchies...

E.g.
```python
@AsyncStatus.wrap
async def completed_status(exception: Optional[Exception] = None) -> None:
    if exception:
        raise exception

def test_my_plan():
    my_device.set = MagicMock(return_value=completed_status())
    my_plan(my_device)
```

DominicOram, 2024-08-16:
Yh, that works, I'm not fussed about the implementation, just the behaviour.

olliesilvester, 2024-08-22:
I can pick this up. @DominicOram what did you mean by having an identifier? Won't it be obvious from whatever test is failing?

DominicOram, 2024-08-27:
It was a request by @dperl-dls. What we've seen in the past is issues where status' are hanging around at the end of tests and have not been sufficiently awaited/cancelled. When this happens it's very hard to tell which of the hundreds of tests produced the status. Personally I think in this case it might not be 100% necessary as it shouldn't really be possible to have the noop hanging around for any length of time.

olliesilvester, 2024-08-27:
I think this part may be addressed in https://github.com/bluesky/ophyd-async/pull/538/files#

DominicOram, 2024-08-27:
In which case with @dperl-dls away in the spirit of keeping things moving I'm happy to drop the requirement of having them named but he's welcome to open a new issue when he's back.

PR #540 (2024-09-03): Add no op status
Fixes #521