bluesky/ophyd-async #527: Make Motor locatable

stan-dot, 2024-08-19:
I think:
- https://github.com/bluesky/bluesky/pull/1536 delivered a `Locatable` protocol and updated the relative set plan to use it
- ophyd-async Motor doesn't support this protocol yet
- ophyd EpicsMotor doesn't support this protocol yet

So that would mean the only thing left to do in this ticket is update ophyd-async Motor and ophyd EpicsMotor to have a `locate()` method that returns the setpoint and readback

_Originally posted by @coretl in https://github.com/bluesky/bluesky/issues/1431#issuecomment-2296161700_

stan-dot, 2024-08-19:
> ERROR: Could not find a version that satisfies the requirement pydantic-numpy (from ophyd-async[dev]) (from versions: none)

branch impacted by the numpy error

stan-dot, 2024-08-23:
wondering
1. when ophyd-async imports bluesky.protocol it's a circular import in a sense
2. both dataclasses and typeddicts are used at the same time...

coretl, 2024-08-23:
>     1. when ophyd-async imports bluesky.protocol it's a circular import in a sense

Why? `ophyd-async` import `bluesky`. `bluesky` does not import `ophyd-async`

>     2. both dataclasses and typeddicts are used at the same time...

Typed dicts to satisfy protocols, dataclasses for things that are internal to ophyd-async

closed 2024-08-27 by stan-dot