bluesky/ophyd-async #532: FastCS Eiger: Allow setting the photon energy and other MX settings

DominicOram, 2024-08-19:
When arming the Eiger you need to provide the photon energy and a number of other metadata parameters (https://github.com/DiamondLightSource/dodal/blob/46132f0e066debaf906c1a80dcefca175b3f5c01/src/dodal/devices/eiger.py#L224). As part of https://github.com/bluesky/ophyd-async/pull/528 the signals were put in to do this but it still needs to be done as part of the `arm` of the eiger.

## Acceptance Criteria
* When arming the eiger the metadata is set

coretl, 2024-08-20:
Can we do this in the plan as `bps.mv` commands rather than putting this in `arm`? I was trying to restrict `prepare` to things that are common to all detectors, that are needed for the triggering system to know about, rather than expanding it to everything

DominicOram, 2024-08-20:
> Can we do this in the plan as `bps.mv` commands rather than putting this in `arm`? I was trying to restrict `prepare` to things that are common to all detectors, that are needed for the triggering system to know about, rather than expanding it to everything

See comment at https://github.com/bluesky/ophyd-async/pull/528#discussion_r1723198852.

coretl, 2025-02-25:
Hold fire on this until after https://github.com/bluesky/ophyd-async/issues/727 is done, as this might be better written as a plan