bluesky/ophyd-async #542: Detector controller disarmed once in `stage`, but armed in both `trigger` and `prepare`

evalott100, 2024-08-27:
Currently, detector controller is only disarmed once:

https://github.com/bluesky/ophyd-async/blob/d7f07488fdfa976037791cbdfdb21e0f7c5f7430/src/ophyd_async/core/_detector.py#L202-L207

However it is armed in both trigger and prepare:

https://github.com/bluesky/ophyd-async/blob/d7f07488fdfa976037791cbdfdb21e0f7c5f7430/src/ophyd_async/core/_detector.py#L253-L257

https://github.com/bluesky/ophyd-async/blob/d7f07488fdfa976037791cbdfdb21e0f7c5f7430/src/ophyd_async/core/_detector.py#L299-L303

This is the cause of all the test failure's [here](https://github.com/bluesky/ophyd-async/pull/538) once we enforce that background tasks are handled before test end.

callumforrester, 2024-08-27:
I may be out of date but I think this is intentional: You should be calling `trigger` in a software scan and `prepare` in a hardware scan, if a plan is calling both then the plan is wrong.

On top of that though I thought `arm()` was idempotent.

coretl, 2024-08-27:
> I may be out of date but I think this is intentional: You should be calling `trigger` in a software scan and `prepare` in a hardware scan, if a plan is calling both then the plan is wrong.

That used to be correct, but then we though that `prepare` would be useful for setting exposure time for software scans

> On top of that though I thought `arm()` was idempotent.

Nope, never has been. The state management was done in `StandardDetector` so it didn't have to be managed in every `DetectorControl` class.

> Thoughts? @coretl 

Don't `caput DET:DRV:Acquire 1` until `trigger` or `kickoff`

## Suggestions to fix

We fix this and https://github.com/bluesky/ophyd-async/issues/405 together.

The addition changes we need here:
- Separate out `DetectorControl.arm(trigger_info)` into `DetectorControl.prepare(trigger_info)` and `DetectorControl.arm()`
- Call `DetectorControl.prepare` in `StandardDetector.prepare` (which is also called in `StandardDetector.trigger` if it hasn't already been called
- Check that trigger_mode is correct then call `DetectorControl.arm` in `StandardDetector.trigger` and `StandardDetector.kickoff`

jwlodek, 2024-08-27:
I ran into this during testing simple scans w/ ophyd-async standard detectors. It manifests as a second frame collected on each `count`. I fixed it locally by removing the arming in prepare, and moving it to explicitly be called after prepare in trigger/kickoff, but haven't had a chance to commit the changes/make a pr

closed 2024-09-13 by ZohebShaikh