bluesky/ophyd-async #550: Logging utils no longer exported

DiamondJoseph, 2024-08-30:


coretl, 2024-08-30:
Which ones in particular?

DiamondJoseph, 2024-09-02:
dodal was using 

```python
from ophyd_async.log import (
    DEFAULT_DATE_FORMAT,
    DEFAULT_FORMAT,
    DEFAULT_LOG_COLORS,
    ColoredFormatterWithDeviceName,
)
from ophyd_async.log import logger as ophyd_async_logger
```

None of these are exported in 0.5.0

ophyd_async_logger was used to configure the logging level of ophyd-async and set it as part of a logging hierarchy
The rest were used to construct a DEFAULT_FORMATTER to make logging consistent between all loggers there (graylog, file, stdout)

DiamondJoseph, 2024-09-02:
While I'm in this issue, other things that dodal was using but that are no longer private:

`_merge_gathered_dicts`, `SoftConverter`,

DiamondJoseph, 2024-09-02:
```python
class ApertureConverter(SoftConverter):
        # Ophyd-async #311 should add a default converter for dataclasses to do this
        def reading(
            self, value: SingleAperturePosition, timestamp: float, severity: int
        ) -> Reading:
            return Reading(
                value=asdict(value),
                timestamp=timestamp,
                alarm_severity=-1 if severity > 2 else severity,
            )
```

SoftConverter was being overriden to allow returning a SingleAperturePosition, a dataclass, which I believe would not actually be allowed as it's not returning one of the types allowed by DataKey? I can only assume this was required previously for it to work and it was working.

```python
@dataclass
class SingleAperturePosition:
    name: str
    GDA_name: str
    radius_microns: float | None
    location: ApertureFiveDimensionalLocation
```

`merge_gathered_dicts` was being used to gather softsignal metadata for devices

coretl, 2024-11-22:
@DiamondJoseph can this be closed?

DiamondJoseph, 2025-02-10:
We're still vendoring in [merge_gathered_dicts](https://github.com/DiamondLightSource/dodal/blob/65edfe249ab220310e7ab7b052a472aa9899f6f0/src/dodal/devices/i22/nxsas.py#L18) to dodal, which I think would be useful to expose. Probably we're re-defining the LOG constants too, which is probably fine.

olliesilvester, 2025-03-14:
Was there discussion on exporting the logger again? In dodal we have https://github.com/DiamondLightSource/dodal/blob/b3cc4d0744a416d48297259c3356039a35719d56/src/dodal/log.py#L17

coretl, 2025-03-14:
I'll happily accept a PR to expose the logger as part of the public API

dan-fernandes, 2025-03-26:
I'm currently re-implementing logging in BlueAPI. I have a feeling that I'm going to run into errors that other people have found and worked around, but currently I'm not fussed about directly interacting with whichever logger dodal/ophyd/bluesky actually use, so don't need them to be public.

I've attached all of BlueAPI's logging handlers to the root logger (`logging.getLogger("")`) which should be the parent of all other loggers. Assuming everyone's sticking to the "no import side effects" rule then there shouldn't be any reason that BlueAPI needs to reset each libraries' loggers' configuration.

I'll update here if I have to change course.