bluesky/ophyd-async #551: Eiger FastCS: Use PVI information to fill in PVs

coretl, 2024-09-02:
As soon as PVI datastructures are published over PVA then this file will drastically change to something like:
```python
class EigerDriverIO(Device):
    bit_depth: SignalR[int]
    stale_parameters: SignalR[bool]
    ...
```
It will also only need to include the signals that are needed at static analysis time, at runtime then all the extras (like `beam_centre_x`, etc) will be filled in automatically.

This will also mean that the class structure and ophyd attribute names will have to exactly match those in PVI.

For this reason I suggest we do this soon, and @GDYendell is going to start looking at this now

_Originally posted by @coretl in https://github.com/bluesky/ophyd-async/pull/528#discussion_r1723187325_

DominicOram, 2024-09-18:
Following https://github.com/bluesky/ophyd-async/issues/557#issuecomment-2357802049 I think we can now do this.

coretl, 2025-02-25:
Steps to complete this:
- Run up the Eiger simulator (https://github.com/DiamondLightSource/fastcs-eiger/tree/readme for getting started and comment on https://github.com/DiamondLightSource/fastcs-eiger/pull/58 if it doesn't Just Work)
- Make `ophyd_async/fastcs/eiger/` and `ophyd_async/fastcs/odin/` modules
- Move `EigerDriverIO` to the new `eiger` module and write in the declarative style as in the issue description
- Move `EigerController` to the new `eiger` module
- Rename `Odin` to `OdinHdfIO` and move to new `odin` module and write in the declarative style
- Move `OdinWriter` to new `odin` module
- Delete `ophyd_async/epics/eiger/` in favour of this new module
- Fix up the tests