bluesky/ophyd-async #553: Is there a way to use IntEnum with enum signals?

canismarko, 2024-09-03:
I've been working on converting some devices to ophyd-async recently. For most enum types I've been doing something like

```python
class Scaler(StandardReadable):
    class CountMode(str, Enum):
        ONE_SHOT = "OneShot"
        AUTO_COUNT = "AutoCount"

    def __init__(self, prefix, name=""):
        ...
        self.count_mode = epics_signal_rw(self.CountMode, f"{prefix}.CONT")
        ...
```

or else using ``SubsetEnum`` to save some keystrokes. I think it mostly makes sense to have these all be string enum types.

However, I also have a handful of PVs that are enums but actually represent boolean states, and will have options like:

- "No" and "Yes"
- "N" and "Y"
- "Disable" and "Enable"

Rather than have separate enum types for each of these, I was hoping to have a single ``BoolEnum`` type for all of them with ``0`` and ``1`` since that's really what they all represent. I can caget and caput integers to these PVs, so I think this is a question of getting ophyd-async (or aioca) to accept an int instead of a str.

I tried using the same pattern as above but with an ``IntEnum`` and get an exception that the type did not match the type of the PV.

## Reasons this would be useful:

1) To make testing the state easier: ``if bool(await sig.get_value()):`` instead of ``if (await sig.get_value()) in ["Yes", "Y", "Enable", ...]:``.
2) Compatibility with the ``PyDMCheckBox`` widget, which expects an ``int`` type. Without this, I have to use an EnumCombobox which ends up with a more cluttered UI layout.
3) Not having to write some variant of ``SubsetEnum(["No", "Yes"])`` over and over.

I'm using aioca==1.7 and ophyd-async==0.5.0, both from PyPI.

coretl, 2024-09-03:
I think we intended `epics_signal_rw(bool, ...)` to work for this so you could `sig,set(True)`, but not sure if we ever got round to implementing this...

canismarko, 2024-09-03:
> but not sure if we ever got round to implementing this...

I think that might have been my first choice, but complained about mismatched types.

Interested in a PR?

Would that be implemented in the backend?

Tom-Willemsen, 2024-09-03:
> I think that might have been my first choice, but complained about mismatched types.

That sounds like https://github.com/bluesky/ophyd-async/issues/516 which I fixed recently and I think is in the latest release?

canismarko, 2024-09-03:
> I think is in the latest release?

So it is. I ran my tests last week and waited to open the issues until after the USA holiday yesterday. Ships passing in the night.

Thanks, @Tom-Willemsen

closed 2024-09-03 by canismarko