bluesky/ophyd-async #564: Config sigs for standard detector and standard flyer should use the same naming, same type

jwlodek, 2024-09-09:
Currently in StandardDetectors we call them `config_sigs` or `_config_sigs`, and internally they are in a `list`. In the StandardFlyer, they are `configuration_signals` and are a `tuple`. We should settle on one or the other for both things.

Also, as mentioned in #563 , for `StandardDetector`s the signals that we typically want in `config_sigs` are instantiated with the detector, meaning they need to be set/updated after the device is initalized. Currently, you can do this with `det._config_sigs = ` or something similar, but we should probably find a way to do this without needing to access the private attribute.

coretl, 2024-09-09:
Do we have a usecase for the ones in `StandardFlyer` yet? Otherwise I'm tempted to delete them until we have a usecase for them

jwlodek, 2024-09-09:
I have not used them personally. I suppose all devices should be able to set their config sigs independently, so these are not really needed?

PR #593 (2024-09-27): Remove config sigs kwarg from flyer since use case for them was unclear.
Resolves #564
closed 2024-09-27 by coretl