bluesky/ophyd-async #578: Make it easy to make a hardware backed soft signal

DominicOram, 2024-09-14:
Often the soft signals on a device may be derived from the value of real hardware signals. As such, when the soft signal is read it should read the real hardware and provide its value based on this. This can be done by having the device optionally provide a readback function when creating the signal.

See https://github.com/DiamondLightSource/dodal/issues/782 for more detail on the usecase and https://github.com/DiamondLightSource/dodal/pull/789 for a solution

Tom-Willemsen, 2024-09-15:
This sounds quite similar to https://github.com/bluesky/ophyd-async/issues/525 , maybe good to add your use-case to that issue as well?

DominicOram, 2024-09-15:
Thanks, I agree, closing this as a duplicate.

closed 2024-09-15 by DominicOram