bluesky/ophyd-async #596: Create a version of AD StandardDetector that allows for using the file plugin's "capture" button to save rather than acquire.

jwlodek, 2024-09-25:
At some of our beamlines, the modus operandi for specific detectors is to run continuously and use the file plugin's capture button or save button (in single mode) to capture individual frames. This is specifically useful for some slow flat panel detectors that need to warm up and stay in acquisition mode to get acceptable noise levels, and with some of our monitoring/diagnostic cameras, where single snapshots are collected for logs.

## Acceptance Criteria
- Class similar to `SingleTriggerDetector` is created that will use either single mode / save or capture mode / num capture = 1 / capture as the trigger
- Tests that check all functionality
- Test with real detector/camera.

coretl, 2024-10-07:
Is this for file-writing detectors?

jwlodek, 2024-10-07:
Yes, specifically we want to integrate a Perkin Elmer flat panel detector. To get acceptable performance, it must run continuously, and we use the "save" button in the AD file plugin as the trigger in the ophyd v1 object.

coretl, 2024-10-07:
I suggest that we do this with `StandardDetector`:
- Make a `PerkinElmerController` that controls both a PerkinElmer and an [NDPluginCircularBuff](https://areadetector.github.io/areaDetector/ADCore/NDPluginCircularBuff.html):
  - On `prepare` (or maybe `stage`?) in internal trigger mode starts the driver in continuous mode and sets the number of triggers on the circular buffer plugin with a buffer of 1 and post-buffer of N-1
  - On `arm` starts the circular buffer plugin going
  - On `disarm` stops the circular buffer
- Then the standard HDF writer plugin should do the right thing

closed 2025-04-04 by jwlodek