bluesky/ophyd-async #603: Kickoff/Complete/Collect crashes with an attribute error when using Internal trigger

jwlodek, 2024-10-07:
When using a standard detector in internal trigger mode, trying to use kickoff/complete/collect, will crash with an attribute error because `self._fly_start` is never set because of this conditional: https://github.com/bluesky/ophyd-async/blob/e291396be1ae795ad153f7c8242243f462cc30e5/src/ophyd_async/core/_detector.py#L331

Is there a reason we need this conditional? Everything works fine if I just remove it. Using kickoff/complete/collect with internal trigger is something I want to do because we want to be able to re-use the same plan for both the `everything is hardware triggered` experiment, and the `everything is timestamped` experiment.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Create standard detector.
2. Prepare in internal trigger mode.
3. Run kickoff/complete/collect
4. See traceback.

## Acceptance Criteria
- Allow for using kickoff/complete collect with internal trigger

ZohebShaikh, 2024-10-07:
This change was made in response to the comment here: https://github.com/bluesky/ophyd-async/pull/568#discussion_r1756674086. @coretl may have more context on the necessity of this requirement.

coretl, 2024-10-07:
We want to be able to parallelize the slow operations of our flyscan:
- Move a motor to the start position (2s)
- Arm the detector (10s)

But actually we might want to do the same flyscan 10 times at different energies, so we decided that we should be able to arm for multiple runs at the start:
- Move motor to start and arm the detector for 10x flyscans (10s)
- Do 1 flyscan
- Move motor to start (2s)
- Do 1 flyscan
- ...

We decided that we should put the arm in `prepare` for this, but actually I'm beginning to think we should put the arm back in `kickoff` after all. Then the first detector `kickoff` of the 10 will arm for 10x flyscans, and subsequent ones will not have to do anything.

@ZohebShaikh I'm not sure if we explored that as an option?

coretl, 2024-11-22:
@jwlodek is this still an issue?

closed 2024-12-10 by coretl