bluesky/ophyd-async #610: Fix types in device_save_loader tests

coretl, 2024-10-14:
At the moment the tests in https://github.com/bluesky/ophyd-async/blob/main/tests/core/test_device_save_loader.py have mismatching types, which will fail after #594 

E.g.
https://github.com/bluesky/ophyd-async/blob/72c61adaee46edae90bac3d7edc034c1a71b084a/tests/core/test_device_save_loader.py#L196
puts a Table to
https://github.com/bluesky/ophyd-async/blob/72c61adaee46edae90bac3d7edc034c1a71b084a/tests/core/test_device_save_loader.py#L33
a `SignalRW[str]`.

Some of this is now duplicated in the "all types" tests:
https://github.com/bluesky/ophyd-async/blob/72c61adaee46edae90bac3d7edc034c1a71b084a/tests/core/test_device_save_loader.py#L70-L91

This ticket is to delete any of the tests that poke Signals with the wrong types, and if they are not covered in the "all types" tests then add functionality to that test for what they are testing.

Note that in a followup ticket the "all types" test will start testing against a real EPICS backend, so this should be a minimal effort ticket to retain the current tests so we know what needs to be ported in the followup ticket.

coretl, 2024-10-14:
yes please

jsouter, 2024-10-14:
scratch that, I guess the issue only arises when the signal is created with str datatype specified and then we try to put to it with a StrictEnum value. We could
- overload __str__ for StrictEnum to return str(self.value) (maybe this is bad for debugging?)
- add in an isinstance(..., StrictEnum) check in PrimitiveSoftConverter to extract the value
- expect that users won't try to do that

coretl, 2024-10-14:
I'm tempted to say "expect that users won't try that", it seems unlikely...

closed 2024-10-23 by coretl