bluesky/ophyd-async #611: Add import-linter

coretl, 2024-10-15:
To avoid making circular imports it would be good to make some rules and add a linter as per https://github.com/DiamondLightSource/mx-bluesky/pull/568 but with the config in a TOML file.

The rules:
- epics depends only on core
- tango depends only on core
- sim depends only on core
- fastcs depends only on core, epics, tango
- plan_stubs can depend on anything

## Acceptance Criteria
- Anything that breaks the above rules errors pre-commit

stan-dot, 2024-11-22:
I assume that we want `core` to be independent now.

should plan_stubs be generic though? separate tango and epics plan stubs?

coretl, 2024-11-22:
> should plan_stubs be generic though? separate tango and epics plan stubs?

Hmm, interesting point, in fastcs we do try/catch imports of epics and tango, and epics works even without the relevant transport, but tango doesn't. This will work at the moment but we might find this breaks in the future.

@abbiemery thoughts?

PR #669 (2024-11-25): Add the python package and the config and make the contracts kept
implemented as discussed in the issue description

Fixes #611
closed 2024-11-25 by coretl