bluesky/ophyd-async #618: Seems like `assert_configuration()` (and presumably also `assert_reading()`) are broken for numpy array types. I guess they probably were already broken before these changes, but one of the tests in dodal uses assert_configuration and I encountered this as I'm currently in the process of going through dodal and mx-bluesky to test out this PR.

coretl, 2024-10-22:
Seems like `assert_configuration()` (and presumably also `assert_reading()`) are broken for numpy array types. I guess they probably were already broken before these changes, but one of the tests in dodal uses assert_configuration and I encountered this as I'm currently in the process of going through dodal and mx-bluesky to test out this PR.
I will keep you posted if I find any other issues.

_Originally posted by @rtuck99 in https://github.com/bluesky/ophyd-async/issues/594#issuecomment-2428651732_
            
I suggest that we move the `assert` functions into a new `ophyd_async.testing` package, make `pytest` a dependency of it, then use `pytest.approx` on the value

coretl, 2024-11-22:
The other alternative is to document that you should use `pytest.approx` in your test code when testing for arrays:
```python
async def test_assert_reading_array(mock_signal_array: SignalRW):
    set_mock_value(mock_signal_array, [1, 2, 4.5])
    dummy_reading = {
        "mock_signal": Reading(
            {"alarm_severity": 0, "timestamp": ANY, "value": pytest.approx([1, 2, 4.5])}
        )
    }
    await assert_reading(mock_signal_array, dummy_reading)
```

@rtuck99 do you prefer moving `ophyd_async.core.asset_*` to `ophyd_async.testing.assert_*` and adding the pytest approx there, or writing docs that tell people to put `pytest.approx` in their code?

coretl, 2024-11-26:
An in between option: do the `pytest` import directly inside `assert_reading`, wrapping any numpy array value with `pytest.approx` and adding `"timestamp": pytest.ANY` and `"alarm_severity": 0` if they don't already exist

coretl, 2024-12-09:
Decided to move things to `ophyd_async.testing` in #695, so this ticket is to add `pytest.approx` to the functions

coretl, 2024-12-09:
@Relm-Arrowny are you interested in taking this on?

Relm-Arrowny, 2024-12-09:
> @Relm-Arrowny are you interested in taking this on?

sure, sound easy enough.

closed 2024-12-10 by coretl