bluesky/ophyd-async #620: StandardDetectors/StreamResource only devices still create empty Events when used in a plan

DiamondJoseph, 2024-10-23:
When running the `count` plan from bluesky.plan_stubs with the `SimPatternDetector` as the only Readable to be collected, I receive StreamResources for the detector's 2 datasets, StreamDatums 2*N (where N is the number of points collected, and 2 is the number of StreamResources) and N Event documents.

The event documents are blank, and if I was running a plan over a very large N very quickly, where StreamDatums are emitted at 10Hz, may overwhelm the total number of messages with uselessness.

```json
{'uid': 'eec7b51b-273f-4732-9e5e-e946e7ab4684', 'time': 1729605127.4888647, 'data': {}, 'timestamps': {}, 'seq_num': 1, 'filled': {}, 'descriptor': '74d4f749-a46f-4aef-8512-6a1a2ed41ba7'}
```
@pytest.fixture
def det(RE: RunEngine, tmp_path: Path) -> StandardDetector:
    with DeviceCollector(mock=True):
        det = PatternDetector(tmp_path / "foo.h5")
    return det

def test_multi_count_output(det: StandardDetector, path_provider, RE):
    docs: dict[str, list[DocumentType]] = {}
    RE(
        count({det}, num=3),
        lambda name, doc: docs.setdefault(name, []).append(doc),
    )
    assert "event" in docs
```

coretl, 2024-11-22:
@DiamondJoseph is this a problem?

DiamondJoseph, 2024-11-22:
It's just inelegant :(

closed 2024-11-22 by DiamondJoseph