bluesky/ophyd-async #624: Make PVI detail DeviceVectors

coretl, 2024-10-25:
Could: Is this there because PVI has a standard that says something like "`thing_n` is part of a list of 'thing's on the device"? Or is it more of a loose convention?

Either way, since we get to decide what PVI says, could we make it return a structure rather than a string to make it clear this is the case? Out of scope for this PR obviously, would be a separate issue.

_Originally posted by @callumforrester in https://github.com/bluesky/ophyd-async/pull/594#discussion_r1812340170_
            

Can do this by using QSRV structures with `[n]` in them:
```
    info(Q:group, {
        "$(IOC_NAME=PANDAQSRV):PVI": {
            "value.pulse[1].d": {
                "+channel": "VAL",
                "+type": "plain"
            }
        }
    })
```

Need to think about if it is even possible to make this backwards compatible...

stan-dot, 2024-11-18:
@coretl not clear what is the use case exactly here. is it just syntactic sugar?

coretl, 2024-11-19:
Fixed by #625

PR #625 (2024-11-01): Make pvi aware of DeviceVector
Fixes #624
closed 2024-11-19 by coretl