bluesky/ophyd-async #636: Change tango decorators to Annotations

coretl, 2024-11-04:
After #598 we should change the tango decorators to match.

E.g.
```python
@tango_polling({"counts": (1.0, 0.1, 0.1), "sample_time": (0.1, 0.1, 0.1)})
class TangoCounter(TangoReadable):
    counts: A[SignalR[int], Format.HINTED_SIGNAL]
    sample_time: A[SignalRW[float], Format.CONFIG_SIGNAL]
    start: SignalX
    reset_: SignalX
```
to
```python
class TangoCounter(TangoReadable):
    # Enter the name and type of the signals you want to use
    # If type is None or Signal, the type will be inferred from the Tango device
    counts: A[SignalR[int], Format.HINTED_SIGNAL, TangoPolling(1.0, 0.1)]
    sample_time: A[SignalRW[float], Format.CONFIG_SIGNAL, TangoPolling(1.0, 0.1)]
    start: SignalX
    reset_: SignalX
```

@burkeds this would allow you to specify:
- per-attribute polling at class definition time, that you could override at class init
- if you don't specify an attribute's polling at class definition time, then you would have to enable it via class init

What are the requirements in terms of polled and non-polled Devices? Do you require the same Ophyd Device to be used with both a polled and a non-polled Tango Server? Also, instead of `TangoPolling(allow_polling, polling_period, abs_change, rel_change)`, should we have a `TangoPolling.SLOW_POLL` and `TangoPolling.FAST_POLL` where the numbers attached to each of those are specified at init or via globals?

## Acceptance Criteria
- `tango_polling` decorator is replaced with `TangoPolling` annotation class

burkeds, 2024-11-04:
Hi Tom. I adding the ability to poll from the ophyd side because Tango servers at DESY have polling disabled by default and there are no plans to support tango servers with polling. As I understand it, this is not typical in the Tango world. A Tango device server with polling enabled may emit events from any number of attributes, so it is possible that some attributes may use Tango events while other attributes will need to be manually polled for the same Tango device server.

It is possible that a device server may be operated in polling or non-polling mode. It is conceivable that one facility may use a non-polling version of a server and another facility a polling version without any other differences in the attributes/commands. 

I would say that it is required that, on initialization, the user can specify which signals if any need to be polled and set the polling rate. Polling is really a signal-level problem more than it is a device-level problem as one does not want to by default poll every signal and risk effectively DDOS'ing the device server. It would be nice if this signal level polling configuration could be done without having to modify the class.

On the subject of SLOW_POLL vs FAST_POLL, this is convenient in setting two polling rates but it doesn't allow for setting change thresholds to trigger callbacks. Is this functionality not desired?

burkeds, 2024-11-04:
@coretl That sounds ok to me. If I were to speculate how community Ophyd devices might be used for Tango, I would guess that most shared devices will not specify polling as I imagine most groups take advantage of Tango events. For groups that need ophyd to poll, they would define a new inheriting class which overwrites signal behavior and adds polling as desired. Your suggestion works well with this speculation.

stan-dot, 2024-11-18:
@coretl could this be excluded from the 1.0 milestone?

coretl, 2024-11-19:
> @coretl could this be excluded from the 1.0 milestone?

No, this is public API, and this isn't a big ticket to do

PR #658 (2024-11-22): Change tango decorators to Annotations
Separated into two commits:
1. Consolidate `ophyd_async.tango.signals` and `ophyd_async.tango.base_devices` into `ophyd_async.tango.core` and remove the imports from the top level `ophyd_async.tango` to make it look like `ophyd_async.epics`
2. Remove the `tango_polling` Device decorator and add a `TangoPolling` annotation to use instead

Fixes #636 

@burkeds please could you comment on whether you agree with 1, and review the code changes for 2 to see if you are happy with it
closed 2024-11-22 by coretl