bluesky/ophyd-async #640: mx-bluesky unit tests taking ~3x longer since signal-typing changes

rtuck99, 2024-11-06:
Since updating mx-bluesky to be compatible with #594, the unit tests are taking significantly longer to complete.
On my desktop machine the difference is ~97s to complete compared to ~27s before

It seems that most of this slowdown is due to waiting in `call_in_bluesky_event_loop()` waiting for `wait_for_connection(coros=device.connect(mock=True))` to complete.

Can this be made faster?

coretl, 2024-11-06:
Can you post a typical test please?

olliesilvester, 2024-11-06:
See https://github.com/DiamondLightSource/mx-bluesky/blob/3845a51a2f083c578b92476702950f4125813b4b/tests/unit_tests/beamlines/i04/test_thawing.py#L118 for a typical test

In our tests, devices are created through fixtures which look like 
`smargon = i03.smargon(fake_with_ophyd_sim=True)`

which in turn calls
`call_in_bluesky_event_loop(...)` and the ophyd-async `wait_for_connection(..)`

DominicOram, 2024-11-19:
I'm happy this is fixed by https://github.com/bluesky/ophyd-async/pull/641

closed 2024-11-19 by DominicOram