bluesky/ophyd-async #643: DeviceVector values don't have the device vector as ``parent``

canismarko, 2024-11-12:
Previously, the ``parent`` attribute of a child device that was part of a ``DeviceVector()`` would be a reference to the DeviceVector itself (i.e. ``assert vector[0].parent is vector``). However, now the child's parent is ``None``.

The recent [new signal typing](https://github.com/bluesky/ophyd-async/commit/a30d8cdedbb1f808d4713c4b2a29a90a73ce99aa#diff-cbd7f3c9eeb42a437f2f38ef9a106cc0009ab3f5b3bea30ff7d9411c00648760) moved the assignment of the ``Device.parent`` attribute from ``Device.set_name`` to ``Device.__setattr__``. In most cases this works, however for ``DeviceVector``s, the individual children are not set as attributes, only a dictionary of the children is set as an attribute, so this assignment of the ``parent`` attribute never happens.

This could be solved by adding something like this to ``DeviceVector.__init__()``:

```python
for child in self._children.values():
    child.parent = self
```

## Steps To Reproduce
Steps to reproduce the behavior:

Here is a test. The last line should pass, but does not because ``my_device.channels[0].parent`` is ``None``.

```python
async def test_device_vector_parent():
    class MyDevice(Device):
        def __init__(self, name):
            self.my_signal = soft_signal_rw(float)
            self.channels = DeviceVector({
                0: soft_signal_rw(float)
            })
            super().__init__(name=name)

    my_device = MyDevice(name="my_device")
    await my_device.connect(mock=True)
    # Good tests
    assert my_device.my_signal.name == "my_device-my_signal"
    assert my_device.my_signal.parent is my_device
    assert my_device.channels.name == "my_device-channels"
    assert my_device.channels.parent is my_device
    assert my_device.channels[0].name == "my_device-channels-0"
    # Bad tests
    assert my_device.channels[0].parent is my_device.channels

```

## Acceptance Criteria
- The above test should pass.

coretl, 2024-11-12:
@canismarko thanks for finding the bug! If should be fixed by #644

closed 2024-11-12 by coretl

canismarko, 2024-11-12:
My pleasure. Thanks for the quick fix!

commit c1db131b (2024-11-12): Set parent of children of DeviceVector passed at init

Fixes #643
commit e00fd083 (2024-11-12): Set parent of children of DeviceVector passed at init (#644)

Fixes #643
commit 040f7a48 (2024-11-14): Set parent of children of DeviceVector passed at init (#644)

Fixes #643