bluesky/ophyd-async #657: Investigate using stricter units on signals

DominicOram, 2024-11-19:
As an MX DAQ developer we often have issues where we get confused about what units physical quantities are being expressed in (e.g. https://github.com/DiamondLightSource/mx-bluesky/issues/220). It may be useful if we could be more strict about this in the code by leveraging something like [pint](https://pypi.org/project/Pint/). This would mean signals are set with `Quantities` that have units and will throw exceptions if the units given in the `EGU` field of the PV do not match. This is a spike to do the investigation on how hard this would be and what an interface would look like.

## Acceptance Criteria
- We investigate using a library like `pint` to allow setting signals with values that have unit metadata
- We investigate how much this would change bluesky etc.
- An estimate for the work is written up, with a mock up of roughly what the interface would look like on the plan/device side
- It is decided whether we do this or not and an ADR is made

coretl, 2025-03-31:
My suggestion would be to:
- Add `pint.facets.numpy.quantity.NumpyQuantity[x]` for x in `int`, `float` and `pint.facets.plain.quantity.PlainQuantity[x]`  for x in the supported numpy numeric types here:
https://github.com/bluesky/ophyd-async/blob/f37968eba275003ac229de5578c7a92971188714/src/ophyd_async/core/_signal_backend.py#L25-L44
- Add support in each `SignalBackend` to deal with type hinted quantity types like `SignalW[PlainQuantity[float]]` or `SignalR[NumpyQuantity[Array1D[np.float64]]]`
- If we see something like this, at runtime we need to ensure it has EGUs and they are valid in the default registry
- At runtime then `get_value()` and `read()` will start returning pint `Quantity` instances
- At runtime `set()` will start requiring pint `Quantity` instances
- The units will be converted before passing through and back to the control system
- We will need to add support for serialization, and decide how the units are placed in the yaml file
- Need to think through the implications of using `Quantity` everywhere when it comes to typing

DominicOram, 2025-10-08:
This obviously all needs a lot more thought but I think I would also like to specify the units when writing the device. This means that I can then reason about them in the rest of the codebase before runtime. In this case, if the units I specify do not match the EGU I would like the device to crash out or shout loudly.

I think just allowing this and not having the generic runtime creation of units would still help a lot in the accidental conversion stuff so if it's easier to do just this first I would go for that

coretl, 2025-10-08:
So would:
```python
self.value = epics_signal_rw(float, prefix + ":VALUE", units="mm")
```
that still gave you a `float` when you `get_value`, but failed loudly at connect if `.EGU` is anything other than "mm" work?

coretl, 2025-10-08:
@Tom-Willemsen what library are you using for units handling?

Tom-Willemsen, 2025-10-08:
> @Tom-Willemsen what library are you using for units handling?

We're using [scipp](https://scipp.github.io/) internally which handles physical units, but also uncertainties, xarray-like abstractions, and integrates nicely with `scippneutron` which has neutron-specific utilities that are useful for us (e.g. "unit conversions" from time-of-flight to wavelength).

yhidaka, 2025-10-29:
In `PAMILA`, I added another layer called `PamilaSignal` on top of an EPICS signal. This new signal handles all device-independent "universal" conversion (e.g., [A] <-> [mA], [mm] <-> [um]) during get/put operations via `pint`. There is one more layer on top of this, where device-dependent "representation" conversions happen (e.g., power supply current in [A] <-> magnetic field kick strength in [mrad]). And then there is another layer on top of it to handle the switching between different targets (e.g., a live machine, internal simulator, external digital twin).

For me, I thought `pint` was perfect. I prefer `my_new_value = self.small_value + self.big_value` to *not* raise any error. As long as the dimensions are correct, it should succeed. If not, `pint` will raise a dimensionality error, which is very nice to catch logical coding errors.

I often have a situation where I need to change an RF frequency whose readback PV is in GHz and setpoint PV is in Hz (I don't know why it's set up this way). For example, I get the current frequency value from the readback first, then add 10 Hz, and finally set the setpoint PV to the new frequency.

What I'm annoyed every time is that I have to double check the exact units of these PVs. I prefer not to think about the details of units (i.e., unit prefixes). With `pint` Quantity objects, all I need is: `delta_freq = Quantity('10 Hz'); new_freq = rb_pv.get() + delta_freq; sp_pv.put(new_freq)`. I don't need to check whether the readback is in Hz or GHz. The `put` method makes sure to convert the newly requested value to be in the correct unit before passing it to the EPICS layer, with something like `new_freq.to(self.raw_unit_str)`, no matter what the unit prefix of the incoming quantity object is.

As another example, we often acquire electron beam orbit. Slow-acquisition PVs return in `mm`, while fast-acquisition PVs return in `nm`. I always cannot remember which is which, but critical to know when computing correction currents for magnets. With `pint` fully integrated, I can just do `orbit.to('m').m` and do matrix calculations with an orbit response matrix after `orbit_reps_mat.to('m/ampere').m`.

By the way, my implementation was based on `ophyd`, not `ophyd-async`. Also, this package was a quick testbed for exploring ideas. So, it only assumed `float`'s (or their arrays), and didn't handle other data types.

yhidaka, 2025-10-31:
- The unit information comes from the configuration file, not directly from the control system, for the reason you mentioned.
- Yes, you'd have to create a Quantity object every time. You can do `Q_('4.2 mm')` or `4.2 * mm` if you define a shortcut like `mm = Q_('1 mm')` beforehand. So, it's not that much work, but will add friction to users, so I didn't push for attaching units to the group, as I thought resistance will be too significant to overcome.

Personally, I would keep the raw signals "as is", and add an additional layer on top of it to handle what you suggested. Although `pint` is very light weight and fast, it still can add noticeable latency when performance matters. So, I think keeping the option for users/developers to directly access the fastest signal layer, bypassing unit attachment, is likely beneficial in some cases. Another reason is that you may want to use the conversion function alone, without getting a new value from a control system.

DominicOram, 2025-11-27:
> So maybe the tools we need are...
> 

Yes, I'm happy with a layer on top. Whilst this approach doesn't address some of what I requested above I'm happy with it as it's a significant improvement on what we have and a lot of the things I requested are controversial/difficult. A layer on top also leaves us open to changing behaviour in the future more easily if we want to.

> Add async QuantityConverter.set(signal: SignalW[float], value: Quantity) that reads units from the signal if not cached, then converts the quantity to those units and sets the signal

I assume this would raise if the units of `value` did not match those of the signal (e.g. setting `eV` to `m` signal) and silently convert setting `m` to a `mm` signal. I am happy with that behaviour.

I think overall the approach sounds good, and it will definitely catch some issues for us, thanks!

mrakitin, 2025-12-02:
A use-case which was mentioned yesterday during the call: a motor record representing a virtual Energy axis ([eV] or [keV]) talking to both a monochromator (changing Bragg angle [deg] or [mrad]/[µrad]) and an undulator (changing the gap [mm]).