bluesky/ophyd-async #666: Naming of signals in ophyd-async is different to ophyd.v1

whs92, 2024-11-21:
In ophyd v1 the names of signals of components of a device are created by joining names with a `_`

so a device like:

```python

class Stage(Device):
    x = Cpt(EpicsMotor, 'X')
    y = Cpt(EpicsMotor, 'Y')

class StageOfStage(Device):
    a = Cpt(Stage, 'A')
    b = Cpt(Stage, 'B')

stage_of_stage = StageOfStage('TEST:', name='stage_of_stage'

```

would have component names like: 

```
stage_a_x
stage_a_y
stage_b_x
stage_b_y
```

In ophyd-async component names are joined with a `-` character instead and any `_` characters are removed from the child name. 
```
            child_name = f"{self.name}-{child_name.strip('_')}" if self.name else ""

```
 https://github.com/bluesky/ophyd-async/blob/main/src/ophyd_async/core/_device.py#L113

If I include `ophyd.v1` and `ophyd-async` devices in a plan together then the signals in event documents from `ophyd-async` and `ophyd.v1` devices look different. 

## Why is this useful?

@coretl 

> The reason it was changed it because given a dash separated name you can split on the dash and walk the tree of devices from the root. With an underscore separated name you cannot. 

@DiamondJoseph suggests it's possible to iterate through the descriptor document to walk the tree. (I've not tried this)

## Why is this a problem?

1. Inconsistency in the signal names in documents is confusing when using ophyd v1 and ophyd-async in the same place
2. The `-` can be interpreted as a -Ve sign in some cases e.g if the signal is monitored in a plan and then you try to access the stream from the run using the databroker client. 

## Proposal

Standardise the signal naming of signals to be the same as in ophyd.v1

coretl, 2024-11-22:
@DominicOram @DiamondJoseph @callumforrester will this break anything if we make `ophyd-async` go back to `_` separated names?

@whs92 I can be convinced on changing `-` to `_` but I still think we should strip underscores, as we sometimes have a signal `read_` which is named like that because otherwise it would shadow `read()`.

Also, in your example, if `x` was named `_x`, would you expect the name to be `stage_a__x` or `stage_a_x`?

callumforrester, 2024-11-22:
Why did we use `-` originally? And did we ever write it down?

coretl, 2024-11-22:
> @coretl
> 
> > The reason it was changed it because given a dash separated name you can split on the dash and walk the tree of devices from the root. With an underscore separated name you cannot.

callumforrester, 2024-11-22:
Sorry, reading failure on my part.

> @DiamondJoseph suggests it's possible to iterate through the descriptor document to walk the tree. (I've not tried this)

Neither have I, but it makes sense to me. 

Are there any other major disadvantages we can think of for going back to `_`? If so, could make it configurable and push the tradeoff to the user? If not, I'm happy to just go with this.

whs92, 2024-11-22:
> Also, in your example, if x was named _x, would you expect the name to be stage_a__x or stage_a_x?

Yes

dperl-dls, 2024-11-22:
Using a character for separation which is not a part of a valid python identifier was a significant improvement on the naming scheme in Ophyd IMO; I would hate to have to go through all our data handling code and rename everything back (but obviously I would be willing to if it's really important).

Is it just the aesthetics of having the different schemes that bothers you @whs92 or are there also functional problems with this?

I would be okay with making it configurable too, but that has some potential pitfalls - it should at least be one global config singleton so that we don't have some devices accidentally unconfigured - and seems like a lot of work.

DiamondJoseph, 2024-11-22:
`"parent-child" == parent.child.name` - if you're going to interrogate your devices based on the documents, without relying on any information passed in the descriptor, then you can replace `-` with `.`

I agree that having a character that isn't a valid Python variable name actively prevents confusion and is a design choice we should stick by- if I have a device named `mirror_heater` with a child `mirror_heater_temp`, I can't distinguish that from a case of having `mirror`, with a child `mirror_heater`.

There must be something that unambiguously cannot be a variable name to allow you to split on it reliably.

whs92, 2024-11-22:
All good points, thanks everyone for your input. 

@dperl-dls 
> Is it just the aesthetics

No,  when you monitor a signal the resulting stream name takes the name of the signal. Using a `-` character there means you can't access the stream using `run.stream-name_monitor` and instead have to use `run["stream-name_monitor"]`. This isn't a breaking problem, but it was confusing.  

I agree with @DiamondJoseph point about the `mirror` and `mirror_heater` example.  However since all ophyd.v1 users are aware of this standard naming construction,  we know not to form this. 

I'd argue that this all might seem minor but trying to reduce differences between ophyd v1 and v2 is helpful to drive adoption. Diamond is maybe the only place where _only_ ophyd async is used.  Most places are using and have been using v1 for many years and are now exploring async. We have mixed environments.

If we leave things as they are I have to write docs for my users explaining why their data contains keys which look different. Aesthetics matter.

whs92, 2024-11-22:
I would also be happy if we make it possible to specify this in ophyd.v1.

My main problem is the inconsistency. I don't really care which one is used, I just want it to be the same,  or be possible to configure it to be the same. 

I agree that the names in ophyd-async are easier to seperate and more likely to be unique in the dataset.

coretl, 2024-11-22:
I suggest we discuss this in the collab meeting on Monday. @whs92 are you around for this?

whs92, 2024-11-22:
Yep. See you there.

DiamondJoseph, 2024-11-25:
Both of the [example devices in happi](https://github.com/pcdshub/happi/blob/master/examples/db.json) are named with underscores in the name- `tv1l0_vgc_1` and `tv1l0_vgc_2`: implying the existence of `tv10` and `tv10_vgc` and suggesting that (at SLAC) the use of underscores in names to represent dashes in the PV prefix is desired/expected behaviour. 

The ophyd docs themselves give `device.cpt_name` as an example.

We can't have a standard based on expected knowledge when knowledge is spread so widely and often each facility is starting from scratch with their own standards and rules (see `dodal`/`happi`/`beacon`...).

PR #675 (2024-11-29): Make name separator configurable
Remove the stripping of underscore while we're at it, this can be done by overriding set_name if required.

Fixes #666