bluesky/ophyd-async #678: PVA backend does not correctly handle long-string waveforms.

jwlodek, 2024-11-26:
When attempting to connect to waveform records with the PVA backend, we get a type error:

```
NotConnected: 
manta_async: NotConnected:
    drv: NotConnected:
        nd_attributes_file: TypeError: XF:31ID1-ES{GigE-Cam:1}cam1:NDAttributesFile with inferred datatype Array1D[np.int8] from typeid='epics:nt/NTScalarArray:1.0' specifier='ab' cannot be coerced to str
    fileio: NotConnected:
        xml_file_name: TypeError: XF:31ID1-ES{GigE-Cam:1}HDF1:XMLFileName_RBV with inferred datatype Array1D[np.int8] from typeid='epics:nt/NTScalarArray:1.0' specifier='ab' cannot be coerced to str
        file_path: TypeError: XF:31ID1-ES{GigE-Cam:1}HDF1:FilePath_RBV with inferred datatype Array1D[np.int8] from typeid='epics:nt/NTScalarArray:1.0' specifier='ab' cannot be coerced to str
        file_name: TypeError: XF:31ID1-ES{GigE-Cam:1}HDF1:FileName_RBV with inferred datatype Array1D[np.int8] from typeid='epics:nt/NTScalarArray:1.0' specifier='ab' cannot be coerced to str
        file_template: TypeError: XF:31ID1-ES{GigE-Cam:1}HDF1:FileTemplate_RBV with inferred datatype Array1D[np.int8] from typeid='epics:nt/NTScalarArray:1.0' specifier='ab' cannot be coerced to str
        full_file_name: TypeError: XF:31ID1-ES{GigE-Cam:1}HDF1:FullFileName_RBV with inferred datatype Array1D[np.int8] from typeid='epics:nt/NTScalarArray:1.0' specifier='ab' cannot be coerced to str
        nd_attributes_file: TypeError: XF:31ID1-ES{GigE-Cam:1}HDF1:NDAttributesFile with inferred datatype Array1D[np.int8] from typeid='epics:nt/NTScalarArray:1.0' specifier='ab' cannot be coerced to str
```

This is because the type provided by PVA is an array of ints, rather than a string.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Install ophyd async with p4p only.
2. Attempt to connect to a waveform signal with FTVL= CHAR.

## Acceptance Criteria
- Can connect to waveform records as strings if FTVL=CHAR when using the PVA backend

jwlodek, 2024-11-26:
Moving from slack to here for visibility.

After some investigation, it does not seem that a pvinfo on the record itself (or the `VAL` field) provides enough information to determine the type - instead we may need to explicitly `pvget` the `.FTVL` field. From here `value.choices[value.index]` will give us what we need. However, to avoid hanging/timeouts when this isn't needed, we need a way to filter signals we need to check this for.

jwlodek, 2024-11-26:
I wonder if the correct approach to fixing this is actually to use `Q:form`, and make corresponding PRs to all of the upstream modules that we want to use this with.

coretl, 2024-11-27:
I think the correct long term thing to do is convert areaDetector to either use `lsi`/`lso` records, or add `Q:form` to the existing waveform records.

In the short term I think putting in the same fix as CA, in `make_converter` if `datatype==str` and PVA is `NDArray[np.int8]` `NTScalarArray` then make a `PvaLongStrConverter` and use that to do the conversion in ophyd