bluesky/ophyd-async #706: Move contents of `sim`

coretl, 2024-12-13:
When writing the docs I realize that the `sim` module isn't really a reasonable simulation of anything, I don't think it should be either. It should either be used to demo (in the tutorials) or in testing (e.g. a OneOfEveryTypeDevice). I suggest we move `ophyd_async.sim.demo` -> `ophyd_async.demo` and `ophyd_async.sim.testing` -> `ophyd_async.testing`. This gives us a nice synergy with the `epics` subpackage:

| Building blocks | Devices for tutorial | For testing core |
|-----------------|----------------------|------------------|
| ophyd_async.core            | ophyd_async.demo                 | ophyd_async.testing          |
| ophyd_async.epics.core      | ophyd_async.epics.demo           | ophyd_async.epics.testing    |
| ophyd_async.tango.core      | ophyd_async.tango.demo           | ophyd_async.tango.testing    |

coretl, 2024-12-13:
@jwlodek @genematx thoughts?

genematx, 2024-12-13:
Sounds sensible to me; thanks for asking, @coretl. I've been always reading `sim` here as "simulated devices for demo purposes".

coretl, 2024-12-16:
Change all the `demo` words to `sim`

PR #708 (2024-12-17): Move contents of `sim`
- `ophyd_async.sim.demo` -> `ophyd_async.sim`
- `ophyd_async.sim.testing` -> `ophyd_async.testing`
- `ophyd_async.*.demo` -> `ophyd_async.*.sim`

Fixes #706 

## Changes
```python
# old
from ophyd_async.sim.demo import SimMotor
# new
from ophyd_async.sim import SimMotor
```
```python
# old
from ophyd_async.epics.demo import Sensor
# new
from ophyd_async.epics.sim import Sensor
```
closed 2024-12-17 by coretl