bluesky/ophyd-async #713: Simulated AreaDetector still in Acquire state after collection when `yield from trigger_and_read(detector, *devices)`

DiamondJoseph, 2024-12-19:
Stacktrace:
[bar.txt](https://github.com/user-attachments/files/18196680/bar.txt)

When calling trigger_and_read rapidly on the SimulatedAreaDetector, backed by the containerised IOC

```sh
ophyd-async --version
0.9.0a1
```

## Steps To Reproduce
Steps to reproduce the behavior:
1. Create a plan that loops over trigger_and_read with a the ADsim
2. Observe that it occasionally (~0.5% of the time?) is still in the Acquire state

## Acceptance Criteria
- The ADSim can reliably be used on scans that collection many, many points without failure.

DiamondJoseph, 2024-12-19:
This was observed by @DiamondRC who can provide the plan if behaviour cannot be replicated

coretl, 2024-12-20:
This looks like a race condition we fixed in Malcolm, but didn't apply here.
Malcolm:
https://github.com/DiamondLightSource/pymalcolm/blob/264ec4a319092a545853f41f89ddf1d21da56504/malcolm/modules/ca/parts/caactionpart.py#L98-L122
Ophyd-async:
https://github.com/bluesky/ophyd-async/blob/c781e27691efde1dd7b4d185252840566b01d041/src/ophyd_async/epics/adcore/_core_logic.py#L101-L107
So if we changed to:
```python
await status
state = None
try:
    async for state in observe_value(driver.detector_state, done_timeout=STATUS_TIMEOUT):
        if state in good_states:
            return
except asyncio.TimeoutError:
    raise ValueError(
        f"Final detector state {state.value} not in valid end "
        f"states: {good_states}"
    )
```
or something similar that should fix it...

canismarko, 2025-04-09:
We've been running into this bug I think with our xspress3 family of detectors. I implemented the suggested solution [in a fork](https://github.com/bluesky/ophyd-async/compare/main...spc-group:ophyd-async:detector_races) and so far we have not triggered this race condition.

coretl, 2025-04-24:
Ok, so this issue becomes making a PR out of https://github.com/bluesky/ophyd-async/compare/main...spc-group:ophyd-async:detector_races with a test where the state updates before or after acquire returns

canismarko, 2025-05-08:
PR is ready. Tests are passing, though the CI is failing because of an unrelated test of sim_motor move profiles.

canismarko, 2025-05-20:
Should be fixed now by https://github.com/bluesky/ophyd-async/pull/889

DiamondJoseph, 2025-05-21:
Fixed by #889

closed 2025-05-21 by DiamondJoseph