bluesky/ophyd-async #721: Rationalize names and behaviour of tutorial devices

coretl, 2025-01-07:
There are a couple of classes of devices that we have in tutorials:
1. Simulations where the logic is in the ophyd level, no control system access required
2. Simulations where the logic is in the EPICS or Tango level, requiring control system access

Although the devices we would like to simulate are similar, the implementations are very different, and the requirement of control system access makes me and @callumforrester think we should make the distinction obvious in the naming.

## Proposal
We should have the following devices:
- `ophyd_async.sim.SimMotor` (existing): ophyd level simulation of a motor
- `ophyd_async.sim.SimPointDetector` and `SimPointDetectorChannel` (new): N counter channels producing integer count rates proportional to a couple of motors
- `ophyd_async.sim.SimFilewritingDetector` (was `PatternDetector`): Gaussian blobs written to HDF with intensity proportional to a couple of motors
- `ophyd_async.epics.demo.DemoMotor` (was `Mover`): as `SimMotor` but with logic in epics database (not an epics simMotor)
- `ophyd_async.epics.demo.DemoPointDetector` and `DemoPointDetectorChannel` (was `SensorGroup` and `Sensor`): as `SimPointDetector` but with logic in epics database
- `ophyd_async.tango.demo.DemoMotor` (was `Mover`): as `SimMotor` but with logic in tango server
- `ophyd_async.tango.demo.DemoPointDetector` and `DemoPointDetectorChannel`: as `SimPointDetector` but with logic in tango server
- `ophyd_async.fastcs.demo.DemoMotor` (was `Mover`): as `SimMotor` but with logic in fastcs server
- `ophyd_async.fastcs.demo.DemoPointDetector` and `DemoPointDetectorChannel`: as `SimPointDetector` but with logic in fastcs server

## Acceptance Criteria
- Files exist with the right functionality

jsouter, 2025-04-01:
I have FastCS Controllers that implement the logic from the epics DemoMotor and DemoPointDetector*, do we want to have tests for these, or just provide documentation for how to spin up the FastCS IOC for different transports?

coretl, 2025-04-01:
> I have FastCS Controllers that implement the logic from the epics DemoMotor and DemoPointDetector*, do we want to have tests for these, or just provide documentation for how to spin up the FastCS IOC for different transports?

If you add the `fastcs.demo` to https://github.com/bluesky/ophyd-async/blob/2984ea49b307b661f8e9c32b66208c0d8da1940e/tests/test_tutorials.py#L30 then that will test they behave in the same way as the epics ones

jsouter, 2025-07-21:
Trying to remember why this stalled, I think I was waiting for main to be fixed in FastCS (wrong controllers were getting passed to callbacks iirc), I will have a look and see what's in my local branch

jsouter, 2025-07-24:
I should be able to get a PR up by next week, looks like I got most of the way in reimplementing the EPICS demo with FastCS Controllers, just trying to figure out some disparity between the expected and calculated motor positions