bluesky/ophyd-async #729: Make it easier to spot when slow tests are added

coretl, 2025-01-10:
We should make sure that all new tests are relatively fast, with a few smoke tests that are allowed to take a bit longer. I suggest a default per-test timeout of 0.5s, with an override for the few tests that take longer than this.

This would involve:
- Setting in `timeout = 0.5` in `[tool.pytest.ini_options]` in `pyproject.ini`
- Adding `@pytest.mark.timeout(<a_longer_timeout>)` for those tests that now start to fail

In a later PR we will make the slow tests faster

DominicOram, 2025-02-27:
Can I suggest we do https://github.com/DiamondLightSource/python-copier-template/issues/235 instead?

DominicOram, 2025-02-27:
Yh, I have no opinion on the implementation. If you want to try out the `pyright` implementation here then put that into the template that's fine. There's no point us putting anything in `mx-bluesky` though if it's going to be part of the template soon anyway

Relm-Arrowny, 2025-05-02:
Is this still needed if so can I pick this one up?

DominicOram, 2025-05-02:
Yh, I think it would be good. Our findings from `mx-bluesky`:
* `pytest-timeout` works well
* We found that some of our tests ran slower in CI, we suspect it's just the GH containers being a little unpredictable, we made the timeout longer in CI than locally to get round this. I think we care more about how long things take locally anyway

Relm-Arrowny, 2025-05-07:
Just to make a note in case I forgot as I am going off for 2 weeks, this issue is completed for Linux and just like mx I have to increase the CI timeout to 2s but local time out is 0.5s, I have add individual timeout for all test that failed locally with timeout = local test time + 2 sec. 
The window CI is failing, this may be due to [this](https://chadgolden.com/blog/github-actions-hosted-windows-runners-slower-than-expected-ci-and-you), I will try to look into it when I get back.

coretl, 2025-05-09:
I'd love it if the windows runners could be made faster, https://ichard26.github.io/blog/2025/03/faster-pip-ci-on-windows-d-drive/ seems to have some tips...

Relm-Arrowny, 2025-06-11:
This has been completed. All local tests now have a default timeout of 0.5s, and any tests that typically fail locally have been given their own longer timeout.

For GitHub Actions, the timeout is set to 2 seconds for Linux and 6 seconds for Windows. We will no longer need a separate, longer timeout for Windows if and when we drop support for Python 3.10.

I also added an extra step to the Windows build to use the faster D: drive. This provided a 10-20% speed increase for those operations, but it was not enough to make a significant difference to the overall build time. 

```YAML
    - name: Set TEMP to D:/Temp
        if: inputs.runs-on == 'windows-latest'
        run: |
          mkdir "D:\\Temp"
          echo "TEMP=D:\\Temp" >> $env:GITHUB_ENV
```

coretl, 2025-06-11:
Please could you make a PR for it, then we can examine the test failure over there...

PR #916 (2025-06-12): 729 make it easier to spot when slow tests are added
fixes #729
closed 2025-06-12 by coretl