bluesky/ophyd-async #742: Introduce `SupersetEnum` to go along with `SubsetEnum` and `StrictEnum`

jwlodek, 2025-01-21:
Add a `SupersetEnum` class that specifies that a corresponding enum signal will have permitted values that are all values of the superset, but not necessarily all of them.

One consideration should also be how to handle empty values, since for example ADAndor uses empty strings "" for enum values for data types it does not support rather than removing them.

## Acceptance Criteria
- `SupersetEnum` exits, and can be used in cases where a Enum record is expected to have a subset of its values. See #724 for an example.
- Make `StrictEnum` and `SubsetEnum` both extend from `Enum`. Currently one is a subclass of the other.

muddi900, 2025-01-21:
If I am getting correctly, you want an Enum subclass, that only accepts a set values for each item. It can accept any of the type, but not some of the types.

If that is correct, I think I can implement it.

coretl, 2025-01-22:
We want 3 Enum subclasses:
```python
class UppercaseNameEnumMeta(EnumMeta):
    def __new__(metacls, *args, **kwargs):
        ret = super().__new__(metacls, *args, **kwargs)
        lowercase_names = [x.name for x in ret if not x.name.isupper()]  # type: ignore
        if lowercase_names:
            raise TypeError(f"Names {lowercase_names} should be uppercase")
        return ret

class StrictEnum(str, Enum, metaclass=UppercaseEnumMeta):
    """All members should exist in the Backend, and there will be no extras"""

class AnyStringUppercaseNameEnumMeta(UppercaseNameEnumMeta):
    def __call__(self, value, *args, **kwargs):  # type: ignore
        if isinstance(value, str) and not isinstance(value, self):
            return value
        return super().__call__(value, *args, **kwargs)

class SubsetEnum(str, Enum, metaclass=AnyStringUppercaseNameEnumMeta):
    """All members should exist in the Backend, but there may be extras"""

class SupersetEnum(str, Enum, metaclass=UppercaseEnumMeta):
    """Some members should exist in the Backend, and there should be no extras"""
```

Anywhere that references `SubsetEnum` in the rest of the codebase should be updated to handle `StrictEnum | SubsetEnum | SupersetEnum` with the relevant checking as per the docstrings above

closed 2025-05-09 by coretl