bluesky/ophyd-async #743: Manual reconnect required after IOC restart

callumforrester, 2025-01-23:
This could be a bug or a feature :) 
We will have to decide how/where we deal with it.

EPICS devices connect to PVs when we use `init_devices`, `ensure_connected` or similar functions that call `.connect` on the device object. They then communicate with these PVs when instructed e.g. via `get_value()`, `set_value()` etc. If the IOC restarts then we currently have to call `.connect` again, otherwise we receive a virtual circuit disconnect error from ChannelAccess.

This behaviour can be confusing from the perspective of users. At DLS, IOCs are automatically restarted when they fall over, meaning that the a user may not even notice that their IOC has restarted between running plans. It then requires some debugging/archaeology to diagnose error above. The fix is then to either call `.connect` manually on all devices or just restart the Python process and start over.

We don't necessarily have to solve this in ophyd-async, but if we don't then we have to implement the fix in many different places (all bluesky applications), so it seems like the logical place to do it from my point of view. Thoughts welcome.

## Steps To Reproduce
Checkout ophyd-async and run the following Python script inside the devcontainer:

```python
import atexit
import subprocess
import sys
from pathlib import Path

import bluesky.plan_stubs as bps
from bluesky.run_engine import RunEngine

import ophyd_async
from ophyd_async.epics.sim import Mover
from ophyd_async.plan_stubs import ensure_connected

PV_PREFIX = "IOC-RESTART-DEMO:"

def start_ioc_subprocess() -> subprocess.Popen:
    """Start an IOC subprocess with EPICS database for sample stage and sensor
    with the same pv prefix
    """

    sim = Path(ophyd_async.__file__).absolute().parent / "epics" / "sim"
    args = [sys.executable, "-m", "epicscorelibs.ioc"]

    # Create X and Y motors
    args += ["-m", f"P={PV_PREFIX}X:"]
    args += ["-d", str(sim / "mover.db")]

    # Start IOC
    process = subprocess.Popen(
        args,
        stdin=subprocess.PIPE,
        stdout=subprocess.PIPE,
        stderr=subprocess.STDOUT,
        universal_newlines=True,
    )
    atexit.register(process.communicate, "exit")
    return process

def demo_restart() -> Mover:
    ioc = start_ioc_subprocess()
    # Uncomment to debug
    # out, err = ioc.communicate()
    # print(out)
    # print(err)

    RE = RunEngine()
    x = Mover(f"{PV_PREFIX}X:")

    # Poke the motor
    RE(ensure_connected(x))
    RE(bps.abs_set(x, 1))
    print("Moved to 1")

    # Restart the IOC
    ioc.kill()
    ioc = start_ioc_subprocess()

    # Poke the motor again, virtual circuit disconnect will be logged
    RE(bps.abs_set(x, 2))
    print("Moved to 2")

    # Reconnect and poke the motor a third time
    RE(ensure_connected(x))
    RE(bps.abs_set(x, 3))
    print("Moved to 3")

if __name__ == "__main__":
    demo_restart()
```

## Acceptance Criteria
- Either...
   - Epics signals automatically attempt to reconnect in the event of a virtual circuit disconnect
   - We agree to push that functionality into our own applications (Hyperion, Blueapi, Queueserver etc.)

coretl, 2025-01-23:
Two issues here:
- Your test script does `abs_set`, which means that it doesn't wait for the move to complete before killing the motor. The virtual circuit disconnect is from the running `mover.set` coro, which fails in the background as expected. Changing to `bps.mv` fixes this. When you do `bps.mv` then `bps.rd` will then read back the correct value afterwards
- After reconnection you attempt to do a connection without waiting for the IOC to actually be ready, sticking a `time.sleep` after the subprocess call or inspecting stdout to see "All initialization complete" makes the reconnect timeout go away.

Was there a beamline issue that prompted this issue?

callumforrester, 2025-01-23:
Yes, red herring, turns out this is the real issue from further testing: https://github.com/bluesky/ophyd-async/issues/744

closed 2025-01-23 by callumforrester