bluesky/ophyd-async #747: SubsetEnum behaviour is unexpected and/or not entirely documented

shihab-dls, 2025-01-23:
It seems that setting a signal with type `SubsetEnum` allows **strings** that are not attributes of the enum. Passing other types, like an **int**, will raise a ValueError
ex: `ValueError: 1 is not a valid TestEnums`

Is this expected behaviour? And if so, should this be better documented in the `SubsetEnum` definition?

## Steps To Reproduce
Steps to reproduce the behavior:

`class TestEnums(SubsetEnum):
    TEST = "str"

class TestDevice(StandardReadable):
    def __init__(self, prefix: str, name: str = ""):
        with self.add_children_as_readables():
            self.signal = epics_signal_rw(TestEnums, f"{prefix}TEST")
        super().__init__(name=name)

async def test_string(RE):
    with DeviceCollector(mock=True):
        my_filter = TestDevice("test-prefix")
    await my_filter.signal.set("str")`

coretl, 2025-01-24:
This is expected behaviour. EPICS enums can be address by string value or integer index, but the integer index is generally more liable to change than the string value. For this reason we make `SubsetEnum` a `str, Enum`, and allow any other `str` to be put. The only exception to this is 2 member enums that are effectively True/False. For these you can use `bool` rather than `SubsetEnum`.

I'm a little confused about the code above, it looks like it should work, does it fail?

shihab-dls, 2025-01-24:
Quick Note: Sorry, I've just edited the issue, as I mistakenly put `signal.set("str")` instead of `signal.set("test_str")` to show any `str` is accepted; although, I think you addressed this in your comment anyways @coretl

olliesilvester, 2025-01-24:
As discussed, setting a `SubsetEnum` signal to an arbitrary string will fail at runtime on real devices, but not in unit tests due to how mock mode works. We should include in a docstring that this behaviour is different in unit tests

closed 2025-02-10 by coretl