bluesky/ophyd-async #751: Testing assert functions don't work with approximate values

olliesilvester, 2025-01-27:
Since the changes in https://github.com/bluesky/ophyd-async/pull/711, tests in Dodal are failing when using functions like `assert_reading` and `assert_configuration`. 

It seems like comparing approximate values no longer works in the same way as it used to:

```
  -         'alarm_severity': <ANY>,
  ?                           ^^^^^
  +         'alarm_severity': 0,
  ?                           ^
  -         'timestamp': <ANY>,
  +         'timestamp': 230.607520706,
  -         'value': 80 ± 8.0e-05,
  ?                    ----------
  +         'value': 80,
```

Using `<ANY>` , as well as comparing values with +/- are no longer working.

See https://github.com/DiamondLightSource/dodal/actions/runs/12988587616/job/36219906783 for more detailed test failures

## Steps To Reproduce
Steps to reproduce the behavior:
1. Clone the dodal branch from the PR: https://github.com/DiamondLightSource/dodal/pull/1014
2. Run tests in `test_undulator`, for example, `test_reading_includes_read_fields`
3. See error

## Acceptance Criteria
- Errors fixed

olliesilvester, 2025-01-27:
Issue rewritten from what I'd originally put since I thought the full diff was actually causing the errors

coretl, 2025-01-28:
I think this is contributing to the problem:
https://github.com/pytest-dev/pytest/issues/3638

A possible fix is to allow the user to omit timestamp and alarm severity from the expected values, and if not present then put the actual values into the expected values so the diff is smaller.

coretl, 2025-02-18:
@jsouter is doing this as part of #723

PR #723 (2025-03-05): Update core and tango tests to match structure of epics tests
Fixes #715
Fixes #751
very WIP, I should rework the tango version of OneOfEverythingDevice to create signals for each type as scalar, spectrum and image, so I can combine test_tango_signals and test_tango_one_of_everything.

Also trying to rework MonitorQueue to use the existing assert_value/assert_reading methods
closed 2025-03-05 by jsouter