bluesky/ophyd-async #753: Can this code be removed?

coretl, 2025-01-28:
Can we remove this?
https://github.com/bluesky/ophyd-async/blob/c867932d8db936fd3eb53a1feef8ccf9b8f0a1f4/src/ophyd_async/epics/adcore/_hdf_writer.py#L68-L69

Doesn't seem to be used anywhere...

## Acceptance Criteria
- Line is removed or explained

jwlodek, 2025-01-28:
This was meant to offer the option of adding the file counter to the end of the hdf file name, but now that I think about it, we reset the counter on closing the writer - we should probably just have the filename provider handle this for us instead.

DiamondJoseph, 2025-02-07:
> we should probably just have the filename provider handle this for us

This is how DLS intend to use it fwiw. Our PathProvider/FilenameProvider doing all of the Diamond filesystem specific handling.

closed 2025-02-07 by jwlodek