bluesky/ophyd-async #754: Update all the detectors and ADCore to match ADAravis

coretl, 2025-01-31:
Change class naming, module naming, add docstrings as per the docs in #700 

## Acceptance Criteria
- Module named as per docs, e.g. `adaravis`
- Class named as per docs, e.g. `AravisController`, `AravisDetector`, `AravisDriverIO`
- Enums prefixed by module name, e.g. `AravisTriggerMode`
- Module has docstring pointing to GitHub repo
- All `IO` classes point to the template they describe in the docs
- All `IO` classes updated to use the declarative style

jwlodek, 2025-01-31:
> All IO classes updated to use the declarative style

Is there a specific reason for going with this over the procedural for these I/O classes?

coretl, 2025-01-31:
> > All IO classes updated to use the declarative style
> 
> Is there a specific reason for going with this over the procedural for these I/O classes?

2 reasons:
- It's shorter
- It makes it easier to support PVI when we put that in areaDetector