bluesky/ophyd-async #757: Possibly delete `TangoReadable`?

coretl, 2025-02-07:
When we make an `EpicsDevice` that is also a `StandardReadable` we inherit from both classes:
https://github.com/bluesky/ophyd-async/blob/ca0ff7f06123ce79ddfd1ffd8727081a2b860af8/src/ophyd_async/epics/demo/_point_detector_channel.py#L17-L21

While in Tango we have a `TangoReadable` baseclass that inherits from `TangoDevice` and `StandardReadable` and does nothing else:
https://github.com/bluesky/ophyd-async/blob/ca0ff7f06123ce79ddfd1ffd8727081a2b860af8/src/ophyd_async/tango/core/_tango_readable.py#L9-L32

And the demo devices inherit from that instead:
https://github.com/bluesky/ophyd-async/blob/ca0ff7f06123ce79ddfd1ffd8727081a2b860af8/src/ophyd_async/tango/demo/_counter.py#L8-L18

I propose that we delete `TangoReadable` and make the Tango Devices inherit from both:
```python
class TangoCounter(StandardReadable, TangoDevice): 
    ...
```

@burkeds what do you think?

## Acceptance Criteria
- TangoReadable is deleted or ADR written to explain why it exists

burkeds, 2025-02-11:
No objections from me.

coretl, 2025-06-09:
yes please

closed 2025-11-12 by coretl