bluesky/ophyd-async #762: Declaring Stream With Collectible and Non-collectible Devices in Nested Plan

dan-fernandes, 2025-02-12:
How?

coretl, 2025-02-12:
The terminology is slightly confusing:
- `collect=False` means call `describe()` to make the descriptor, then `read()` to make the event document
- `collect=True` means call `describe_collect()` to make the descriptor then `collect()` to make the event documents

In both cases `collect_asset_docs()` is called, so file writing detectors are supported in the same way in both modes.

dan-fernandes, 2025-02-17:
For context, the [plan](https://github.com/DiamondLightSource/dodal/blob/a80d326193eb7f3cb1879d3c0697bb47d56fa1e6/src/dodal/plans/bimorph.py#L101) is [here](https://github.com/DiamondLightSource/dodal/blob/a80d326193eb7f3cb1879d3c0697bb47d56fa1e6/tests/plans/test_bimorph.py#L637) and the specific test is here. A snippet of my inner run is:
```def inner_scan(
    detectors: list[Readable],
    mirror: BimorphMirror,
    slits: Slits,
    active_dimension: SlitDimension,
    active_slit_center_start: float,
    active_slit_center_end: float,
    active_slit_size: float,
    number_of_slit_positions: int,
    run_metadata: dict[str, Any] | None = None,
):
    """Inner plan stub, which moves Slits and performs a read.

    Args:
        mirror: BimorphMirror to move
        slit: Slits
        oav: oav on-axis viewer
        active_dimension: SlitDimension that slit will move in (X or Y)
        active_slit_center_start: float start position of center of slit in active dimension
        active_slit_center_end: float final position of center of slit in active dimension
        active_slit_size: float size of slit in active dimension
        number_of_slit_positions: int number of slit positions per pencil beam scan
        run_metadata: Optional dict[str, Any] to add as metadata to run start
    """
    yield from bps.open_run(run_metadata)

    yield from bps.declare_stream(*detectors, name="detectors", collect=True)
    yield from bps.declare_stream(mirror, slits, name="primary")

    for value in linspace(
        active_slit_center_start, active_slit_center_end, number_of_slit_positions
    ):
        yield from move_slits(slits, active_dimension, active_slit_size, value)
        yield from bps.trigger_and_read(detectors, name="detectors")
        yield from bps.trigger_and_read((mirror, slits), name="primary")

    yield from bps.close_run()
```

This is wrapped by an outer run, which stages the devices and sets a run key decorator.

I am using nested runs in a plan. In each inner run, I am reading from a series of `Readable`s and `Movable`s. One such `Readable` is the `PatternDetector`. In the inner run, I call ```yield from bps.declare_stream(*detectors, name="detectors", collect=True``` where detectors contains my `PatternDetectors`.
This gives a:
```AssertionError: `declare_stream` contained `collect=True` but  `describe_collect` did not return a single Dict[str, DataKey] for the passed in detectors```
I think this is because `PatternDetector` inherits from `StandardDetector` which populates its `StandardDetector._describe` attribute in `StandardDetector.prepare` method. As this is not called before declaring the stream, `PatternDetector.describe_collect` does not return a usable dictionary.

Do you have any idea if this is an issue with the detectors, or with my plan?

coretl, 2025-02-17:
`StandardDetector` supports either:
- The traditional "deferred" bluesky approach where the first time you `read()` it calls `describe()` if not already done so. In this case you don't call `declare_stream` at all, and the `prepare` is done implicitly by `trigger()`
- The "up front" approach where you call `declare_stream` before calling `read()`. If you do this then you need to have called `prepare()` before `declare_stream`.

I suggest you either:
- delete `declare_stream` and let it be deferred
- or do a `prepare()` before the `declare_stream`

Also with `trigger_and_read` you need `collect=False`. And is there are reason the mirrors, slits and detectors aren't in the same stream?

It occurs to me that none of this is well documented, so I'll have a go at improving the docs when I write the detectors tutorial

dan-fernandes, 2025-02-17:
The original inner run was as follows:
```def inner_scan(
    detectors: list[Readable],
    mirror: BimorphMirror,
    slits: Slits,
    active_dimension: SlitDimension,
    active_slit_center_start: float,
    active_slit_center_end: float,
    active_slit_size: float,
    number_of_slit_positions: int,
    run_metadata: dict[str, Any] | None = None,
):
    """Inner plan stub, which moves Slits and performs a read.

    Args:
        mirror: BimorphMirror to move
        slit: Slits
        oav: oav on-axis viewer
        active_dimension: SlitDimension that slit will move in (X or Y)
        active_slit_center_start: float start position of center of slit in active dimension
        active_slit_center_end: float final position of center of slit in active dimension
        active_slit_size: float size of slit in active dimension
        number_of_slit_positions: int number of slit positions per pencil beam scan
        run_metadata: Optional dict[str, Any] to add as metadata to run start
    """
    yield from bps.open_run(run_metadata)

    for value in linspace(
        active_slit_center_start, active_slit_center_end, number_of_slit_positions
    ):
        yield from move_slits(slits, active_dimension, active_slit_size, value)
        # yield from bps.trigger_and_read(detectors, name="detectors")
        # yield from bps.trigger_and_read((mirror, slits), name="primary")
        yield from bps.trigger_and_read((*detectors, mirror, slits))

    yield from bps.close_run()
```

This gives the error:
```
...
  File "/venv/lib/python3.11/site-packages/bluesky/bundlers.py", line 833, in _pack_external_assets
    data_keys_received.add(self._stream_resource_data_keys[doc["stream_resource"]])
                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: '0b4747e7-947a-4b7e-aec2-3ada3946e8f8'
```

We put in stream declarations as the nested run documentation seemed to imply it may be necessary.

We split the detectors and other devices into two streams as we wrongly assumed one would need `collect=True` and other `=False`.

When are declaring streams required?

coretl, 2025-02-17:
Ah, this looks like it might be usecase 3 from:
https://github.com/bluesky/bluesky/issues/1849

The stage is on the outer plan (which implicitly does a prepare on the first trigger) so the detectors are trying to use the same file for the entire outer plan.

The question is, do you really want use case 3? Or are you after use case 1 or 2. If you want use case 1 then move the run start and end into the outer plan. If you want use case 2 then put a prepare in the inner plan.

dan-fernandes, 2025-02-18:
Having added a `bps.prepare` to the inner scan I am still getting a stream resource key-error.

As @DominicOram suggested my inner plan is wrapper in a `set_run_key_wrapper`:
```
...
            inner = inner_scan(
                detectors,
                mirror,
                slits,
                active_dimension,
                active_slit_center_start,
                active_slit_center_end,
                active_slit_size,
                number_of_slit_positions,
                run_metadata={
                    # "run_key": f"inner_key_{i + 1}",
                    "outer_uid": outer_uid,
                    "bimorph_position_index": i + 1,
                },
            )
            yield from bpp.set_run_key_wrapper(inner, f"inner_key_{i + 1}")
...
```

Inner:
```
def inner_scan(
    detectors: list[Readable],
    mirror: BimorphMirror,
    slits: Slits,
    active_dimension: SlitDimension,
    active_slit_center_start: float,
    active_slit_center_end: float,
    active_slit_size: float,
    number_of_slit_positions: int,
    run_metadata: dict[str, Any] | None = None,
):
    """Inner plan stub, which moves Slits and performs a read.

    Args:
        mirror: BimorphMirror to move
        slit: Slits
        oav: oav on-axis viewer
        active_dimension: SlitDimension that slit will move in (X or Y)
        active_slit_center_start: float start position of center of slit in active dimension
        active_slit_center_end: float final position of center of slit in active dimension
        active_slit_size: float size of slit in active dimension
        number_of_slit_positions: int number of slit positions per pencil beam scan
        run_metadata: Optional dict[str, Any] to add as metadata to run start
    """
    yield from bps.open_run(run_metadata)

    for detector in detectors:
        detector.prepare(TriggerInfo(number_of_triggers=1))

    for value in linspace(
        active_slit_center_start, active_slit_center_end, number_of_slit_positions
    ):
        yield from move_slits(slits, active_dimension, active_slit_size, value)
        yield from bps.trigger_and_read([*detectors, mirror, slits])

    yield from bps.close_run()```
```
Error:

```
  File "/venv/lib/python3.11/site-packages/bluesky/run_engine.py", line 2076, in _save
    await current_run.save(msg)
  File "/venv/lib/python3.11/site-packages/bluesky/bundlers.py", line 545, in save
    indices_generated = await self._pack_external_assets(self._asset_docs_cache, message_stream_name=desc_key)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/venv/lib/python3.11/site-packages/bluesky/bundlers.py", line 833, in _pack_external_assets
    data_keys_received.add(self._stream_resource_data_keys[doc["stream_resource"]])
```

I am not sure what to try next.

coretl, 2025-02-18:
Could you confirm whether the error happens on the first or second call of inner_scan?

dan-fernandes, 2025-02-18:
I've re-written the code to use no nested runs, by removing the outer runs.

The inner plan still creates new a new run every time it is executed.

This gives the same error. I have checked, and it occurs on the second execution.

dan-fernandes, 2025-02-18:
Having checked out a previous commit where nested runs were still present, the exception is on the second inner run.

dan-fernandes, 2025-02-18:
This is my new inner plan:
```
def inner_scan(
    detectors: list[Readable],
    mirror: BimorphMirror,
    slits: Slits,
    active_dimension: SlitDimension,
    active_slit_center_start: float,
    active_slit_center_end: float,
    active_slit_size: float,
    number_of_slit_positions: int,
    run_metadata: dict[str, Any] | None = None,
):
    """Inner plan stub, which moves Slits and performs a read.

    Args:
        mirror: BimorphMirror to move
        slit: Slits
        oav: oav on-axis viewer
        active_dimension: SlitDimension that slit will move in (X or Y)
        active_slit_center_start: float start position of center of slit in active dimension
        active_slit_center_end: float final position of center of slit in active dimension
        active_slit_size: float size of slit in active dimension
        number_of_slit_positions: int number of slit positions per pencil beam scan
        run_metadata: Optional dict[str, Any] to add as metadata to run start
    """
    yield from bps.open_run(run_metadata)
    yield from bps.stage_all(*detectors, mirror, slits)

    # for detector in detectors:
    #     detector.prepare(TriggerInfo(number_of_triggers=1))

    for value in linspace(
        active_slit_center_start, active_slit_center_end, number_of_slit_positions
    ):
        yield from move_slits(slits, active_dimension, active_slit_size, value)
        yield from bps.trigger_and_read([*detectors, mirror, slits])

    breakpoint()

    yield from bps.unstage_all(*detectors, mirror, slits)
    yield from bps.close_run()
```