bluesky/ophyd-async #779: Mixed signal implementation conflicting with _device_filler.

burkeds, 2025-02-20:
I have a tango server which can be simulated or not. If simulated, trying to read PositionEncoder throws an error. I want to be able to conditionally create this signal based on the `sim` argument in `__init__`. The problem is, when connecting, the `_device_filler` throws an exception because it won't allow itself to overwrite the existing child.

Should we allow mixed implementation like this? If so, should the _device_filler not try to create a child with the same name as an existing child?

```python
class OmsVME58Motor(FSECReadableDevice, Movable, Stoppable):
    Position: A[SignalRW[float], Format.HINTED_UNCACHED_SIGNAL]

    def __init__(
            self,
            trl: str | None = None,
            device_proxy: DeviceProxy | None = None,
            name: str = "",
            sim: bool = False,
    ) -> None:
        if not sim:
            with self.add_children_as_readables(Format.UNCACHED_SIGNAL):
                self.PositionEncoder = tango_signal_r(float, trl+"/PositionEncoder", device_proxy)
                self.PositionEncoderRaw = tango_signal_r(float, trl+"/PositionEncoderRaw", device_proxy)
        super().__init__(trl=trl, device_proxy=device_proxy, name=name)
        self._set_success = False
```

```
File "/home/burkedev/projects/BLUESKY/P65/venv/lib/python3.11/site-packages/ophyd_async/core/_device_filler.py", line 92, in _raise
    raise TypeError(f"{type(self._device).__name__}.{name}: {error}")
TypeError: OmsVME58Motor.PositionEncoder: Cannot make child as it would shadow <ophyd_async.core._signal.SignalR object at 0x7fa61aed2250>
```

coretl, 2025-02-20:
A few questions:
- In sim mode does the Attribute actually exist, or it exists but errors?
- Do you use PositionEncoder in any code apart from adding it above to `StandardReadable`?

You could solve it at present by splitting into a `class OmsVME58MotorSim(TangoDevice)` and a `class OmsVME58Motor(OmsVME58MotorSim)`. 

I didn't add a way for DeviceFiller to validate existing Signals, but that could be added with a bit of thought

burkeds, 2025-02-21:
>     * In sim mode does the Attribute actually exist, or it exists but errors?

- There is no difference in attributes between simulated and non-simulated modes. PositionEncoder exists so when I read the device, the device server throws an exception when the attribute is read.

>     * Do you use PositionEncoder in any code apart from adding it above to `StandardReadable`?

- I have this 0msVME58Motor as a general purpose motor on the beamline. I have a production monochromator with three of them and a host of other real and simulated motors. PositionEncoder needs to be read on the monochromator and I would like to use the same device classes when testing using simulated motors and in production deployments.

Of course I could write a new class for simulated motors. In the Tango world, there can be variation in tango servers controlling the same physical device, even within the same facility. Permitting mixed implementations like this would allow for more flexibility to accommodate this variance.

burkeds, 2025-02-27:
> You could solve it at present by splitting into a `class OmsVME58MotorSim(TangoDevice)` and a `class OmsVME58Motor(OmsVME58MotorSim)`.

I don't think this is a solution either. There is no way to exclude signals from or disable the signal filler when instantiating a TangoDevice. So long as two TangoDevices have the same TRL, they will share the same autofilled signals.

This is a problem for my use case but I see another issue.

Many device servers are sort of "hacked together" with years of changes and in some cases, vestigial attributes which are unused or only used with specific hardware configurations. If any of these attributes throw an exception when read, say when `store_settings` is run, then that becomes a problem.

Here are some potential solutions as I see it.

- Add an init argument to disable device filling so only annotated signals are connected. Not sure whether or not to put this in TangoDevice or Device. I thought I remembered creating something like this for Tango but I don't know if it survived into the current version.
- Create a signal annotation which is the equivalent of "ignore this signal" such that it is explicitly not created/connected/auto-filled

closed 2025-03-03 by coretl